Xilinx/RecoNIC #5: What's the mapping status between ERNIC interrupts and MSI-X interrupts on RecoNIC

archerypp, 2023-10-16:
Dear Zhguanw,
  I would like to have a try to port U250 kernel verbs driver on RecoNIC. Then I meet a problem that how to get the interrupts of RecoNIC for PCIe host driver. I believe there should be some MSI-X interrupts. But I am not sure if RecoNIC already supports mapping ERNIC interrupts to MSI-x. Could you please help with this?

zhguanw-amd, 2023-10-17:
Hi Archer,

Thanks for your interests.

QDMA in RecoNIC shell has the MSI-X interrupts. However, at the current implementation, ERNIC's interrupt is not used and there is no connection between ERNIC's interrupt signal and QDMA's interrupt interface. For the host side, we rely on polling at the moment. We are working on a version that accelerators can access ERNIC also, which means that doorbells ringing/polling can be done inside accelerators as well.

If you could contribute the interrupt feature to RecoNIC infrastructure, that would be great to the community.

Thanks,
Guanwen

archerypp, 2023-10-21:
Hi Guanwen,
  Understood. Thanks for your infomation.
Best Regards
Archer

closed 2023-10-21 by archerypp