Xilinx/RecoNIC #10: Try porting the project to VCU128

niexun725, 2023-12-10:
Dear Zhguanw,
    I am trying to port the project to two Virtex UltraScale + HBM VCU128 FPGA, the system version is Ubuntu 20.04 and linux kernel version 5.4.0-125-generic, and the vivado version is also 2021.2. Everything looks fine until I run the RDMA Read test，Host and Client successfully established a connection, but the Client shows Error : received data mismatched. 
    I tried to use ila to monitor the AXI4 interface of dev ddr4, and found that the host successfully wrote the data to the DDR4 to be read by the client, and finally the client also read the corresponding amount of data from its own DDR4.However, the client did not actually read the DDR4 of the Host successfully, because I did not see the read operation on the Host DDR4. By the way, the client shows "Successfully send an RDMA read operation"
    I guess the reason is the DDR4 capacity of VCU128 DDR4 is too small, only 4GB, even if its chip architecture is almost the same with AU280 , some operations make the memory boundary overflow, I would like to ask if RecoNIC has some limitations on the capacity of DDR？I am learning this project, hoping to get your guidance.
Thanks,
Xun Nie

zhguanw-amd, 2023-12-11:
Hi Xun Nie,

Thanks for your interests and the efforts on porting RecoNIC to VCU128!

From your description, it seems that the client can successfully send RDMA read request packet to the server and the server also can successfully generate RDMA read response packet to the client. The issue happens when the client copies data from the device memory to host memory for verification via the QDMA AXI-MM channel.

Could you reprogram the FPGA, run your read test again and dump RDMA registers? It would be helpful if you can post your dumped RDMA register here after you run read operation. Before you run the read test, could you also share your dmesg information related to onic-driver as well?

>I guess the reason is the DDR4 capacity of VCU128 DDR4 is too small, only 4GB, even if its chip architecture is almost the same with AU280 , some operations make the memory boundary overflow, I would like to ask if RecoNIC has some limitations on the capacity of DDR？

I don't think this is related to DDR capacity. RecoNIC defines the device memory size at https://github.com/Xilinx/RecoNIC/blob/main/lib/reconic.h#L47, but this can be configurable, as long as we can instantiate more DDRs in the hardware. Probably you can check whether addresses of the AXI read transaction from host to the device memory after receiving RDMA read response packet at the client side via ILA are correct or not.

Thanks,
Guanwen

niexun725, 2023-12-11:
Thanks for your reply! I have reprogrammed the FPGA，and got the following results. As “Info: [RN_RDMA_GCSR_ERRBUFWPTR      = 0x6006c] = 0x2” shows, the host received some error packet. 

Besides, After use ila, for the box_250mhz module of host, the"a_axis_adap_rx_250mhz*"interface, it indeed has some input packets. However, "m_axis_user2rdma_roce_from_cmac_rx_* " interface nerver asserts. I think packet classification doesn't work, but I indeed have a full license of vitis_network_p4.

Here are some of the results displayed after the console runs the dmesg and rdma read instructions on both client and host ( too much content, sorry to take up your time )

After running "dmesg" On the client side:
```
[   39.512468] onic: loading out-of-tree module taints kernel.
[   39.513196] onic: module verification failed: signature and/or required key missing - tainting kernel
[   39.517258] onic 0000:01:00.0 onic1s0f0 (uninitialized): Set MAC address to 0:a:35:1b:40:13
[   39.517258] onic 0000:01:00.0: device is a master PF
[   39.517260] onic_set_num_queue: num_msix 10, nb_queues 9, pci_msix_user_cnt 1
[   39.517261] onic_pci_probe mm_queues: 4
[   39.517293] onic:qdma_device_open: onic, 01:00.00, pdev 0x00000000258ad560, 0x10ee:0x903f.
[   39.517299] onic 0000:01:00.0: enabling device (0100 -> 0102)
[   39.517402] Device Type: Soft IP
[   39.517403] IP Type: EQDMA Soft IP
[   39.517403] Vivado Release: vivado 2020.2
[   39.517407] onic:qdma_device_attributes_get: qdma01000-p0000:01:00.0: num_pfs:1, num_qs:512, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[   39.517408] onic:qdma_device_open: Vivado version = vivado 2020.2
[   39.517409] qdma_dev_entry_create: Created the dev entry successfully
[   39.520710] onic:xdev_identify_bars: AXI Master Lite BAR 2.
[   39.520712] onic:qdma_device_open: 0000:01:00.0, 01000, pdev 0x00000000258ad560, xdev 0x000000004e84b042, ch 1, q 1024, vf 0.
[   39.622831] onic 0000:01:00.0 enp1s0: renamed from onic1s0f0
[   39.647248] real_num_tx_queues: 68 real_num_rx_queues 68
[   39.647266] onic 0000:01:00.0: onic_cdev_ptr->name = reconic-mm
[   39.647269] onic 0000:01:00.0: successfully cdev_add a character device, reconic-mm, to the system
[   39.647949] onic 0000:01:00.0: successffully device_create a character device, reconic-mm, and register it
[   39.661583] onic 0000:01:00.0 enp1s0: onic_open: device open done
```
After running "dmesg" On the host side:
```
[  160.012775] onic: loading out-of-tree module taints kernel.
[  160.012935] onic: module verification failed: signature and/or required key missing - tainting kernel
[  160.015234] onic 0000:04:00.0 onic4s0f0 (uninitialized): Set MAC address to 0:a:35:94:71:bf
[  160.015235] onic 0000:04:00.0: device is a master PF
[  160.015244] onic_set_num_queue: num_msix 10, nb_queues 9, pci_msix_user_cnt 1
[  160.015244] onic_pci_probe mm_queues: 4
[  160.015284] onic:qdma_device_open: onic, 04:00.00, pdev 0x000000001914674f, 0x10ee:0x903f.
[  160.015298] onic 0000:04:00.0: enabling device (0000 -> 0002)
[  160.015500] Device Type: Soft IP
[  160.015501] IP Type: EQDMA Soft IP
[  160.015501] Vivado Release: vivado 2020.2
[  160.015509] onic:qdma_device_attributes_get: qdma04000-p0000:04:00.0: num_pfs:1, num_qs:512, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[  160.015510] onic:qdma_device_open: Vivado version = vivado 2020.2
[  160.015512] qdma_dev_entry_create: Created the dev entry successfully
[  160.022109] onic:xdev_identify_bars: AXI Master Lite BAR 2.
[  160.022112] onic:qdma_device_open: 0000:04:00.0, 04000, pdev 0x000000001914674f, xdev 0x00000000815ae9a2, ch 1, q 1024, vf 0.
[  160.141314] onic 0000:04:00.0 enp4s0: renamed from onic4s0f0
[  160.205177] real_num_tx_queues: 68 real_num_rx_queues 68
[  160.205194] onic 0000:04:00.0: onic_cdev_ptr->name = reconic-mm
[  160.205195] onic 0000:04:00.0: successfully cdev_add a character device, reconic-mm, to the system
[  160.205351] onic 0000:04:00.0: successffully device_create a character device, reconic-mm, and register it
[  160.243862] onic 0000:04:00.0 enp4s0: onic_open: device open done
[  161.149484] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
```
After run rdma read test on the client:
```
root@niexun-HP-288-Pro-G6-Microtower-PC:/home/niexun/RecoNIC/examples/rdma_test# sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./read -r 192.100.51.1 -i 192.100.52.1 -p /sys/bus/pci/devices/0000:01:00.0/resource2 -z 128 -l dev_mem -d /dev/reconic-mm -c -u 22222 -t 11111 --dst_qp 2 -g 2>&1 | tee client_debug.log
src_ip_str = 192.100.51.1
dst_ip_str = 192.100.52.1
Info: PCIe resource file: /sys/bus/pci/devices/0000:01:00.0/resource2
Info: QP allocated at: dev_mem
Info: Device - /dev/reconic-mm
Info: src_ip = 192.100.51.1
Info: Found network interface: enp1s0
Info: mac_addr_t = 00:0a:35:1b:40:13
Info: Creating rn_dev
/home/niexun/RecoNIC/lib/reconic.c:296:create_rn_dev(): Info: scr(=4)) file open successfully
[... 367 lines cut ...]
Info: [RN_RDMA_GCSR_CQINTSTS4       = 0x601bc] = 0x0
Info: [RN_RDMA_GCSR_CQINTSTS5       = 0x601c0] = 0x0
Info: [RN_RDMA_GCSR_CQINTSTS6       = 0x601c4] = 0x0
[... cut ...]
/home/antl/RecoNIC/lib/rdma_api.c:1087:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f3080c59000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/home/antl/RecoNIC/lib/rdma_api.c:1095:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f3080c59000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
[1]    20813 segmentation fault  sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./read -r 192.100.52.1 -i  -p  -z   | 
       20814 done                tee server_debug.log
```

Really appreciate for your guidance！
Xun Nie

niexun725, 2023-12-11:
> Could you reprogram the FPGA, run your read test again and dump RDMA registers? It would be helpful if you can post your dumped RDMA register here after you run read operation. Before you run the read test, could you also share your dmesg information related to onic-driver as well?

@zhguanw-amd 
Thanks for your reply! I have reprogrammed the FPGA，and got the following results. As “Info: [RN_RDMA_GCSR_ERRBUFWPTR      = 0x6006c] = 0x2” shows, the host received some error packet. 

Besides, After use ila, for the box_250mhz module of host, the"a_axis_adap_rx_250mhz*"interface, it indeed has some input packets. However, "m_axis_user2rdma_roce_from_cmac_rx_* " interface nerver asserts. This might means that no packets are identified as rdma traffic on the host.I think packet classification doesn't work, but I indeed have a full license of vitis_network_p4.

Here are some of the results displayed after the console runs the dmesg and rdma read instructions on both client and host ( too much content, sorry to take up your time )

After running "dmesg" On the client side:
```
[   39.512468] onic: loading out-of-tree module taints kernel.
[   39.513196] onic: module verification failed: signature and/or required key missing - tainting kernel
[   39.517258] onic 0000:01:00.0 onic1s0f0 (uninitialized): Set MAC address to 0:a:35:1b:40:13
[   39.517258] onic 0000:01:00.0: device is a master PF
[   39.517260] onic_set_num_queue: num_msix 10, nb_queues 9, pci_msix_user_cnt 1
[   39.517261] onic_pci_probe mm_queues: 4
[   39.517293] onic:qdma_device_open: onic, 01:00.00, pdev 0x00000000258ad560, 0x10ee:0x903f.
[   39.517299] onic 0000:01:00.0: enabling device (0100 -> 0102)
[   39.517402] Device Type: Soft IP
[   39.517403] IP Type: EQDMA Soft IP
[   39.517403] Vivado Release: vivado 2020.2
[   39.517407] onic:qdma_device_attributes_get: qdma01000-p0000:01:00.0: num_pfs:1, num_qs:512, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[   39.517408] onic:qdma_device_open: Vivado version = vivado 2020.2
[   39.517409] qdma_dev_entry_create: Created the dev entry successfully
[   39.520710] onic:xdev_identify_bars: AXI Master Lite BAR 2.
[   39.520712] onic:qdma_device_open: 0000:01:00.0, 01000, pdev 0x00000000258ad560, xdev 0x000000004e84b042, ch 1, q 1024, vf 0.
[   39.622831] onic 0000:01:00.0 enp1s0: renamed from onic1s0f0
[   39.647248] real_num_tx_queues: 68 real_num_rx_queues 68
[   39.647266] onic 0000:01:00.0: onic_cdev_ptr->name = reconic-mm
[   39.647269] onic 0000:01:00.0: successfully cdev_add a character device, reconic-mm, to the system
[   39.647949] onic 0000:01:00.0: successffully device_create a character device, reconic-mm, and register it
[   39.661583] onic 0000:01:00.0 enp1s0: onic_open: device open done
```
After running "dmesg" On the host side:
```
[  160.012775] onic: loading out-of-tree module taints kernel.
[  160.012935] onic: module verification failed: signature and/or required key missing - tainting kernel
[  160.015234] onic 0000:04:00.0 onic4s0f0 (uninitialized): Set MAC address to 0:a:35:94:71:bf
[  160.015235] onic 0000:04:00.0: device is a master PF
[  160.015244] onic_set_num_queue: num_msix 10, nb_queues 9, pci_msix_user_cnt 1
[  160.015244] onic_pci_probe mm_queues: 4
[  160.015284] onic:qdma_device_open: onic, 04:00.00, pdev 0x000000001914674f, 0x10ee:0x903f.
[  160.015298] onic 0000:04:00.0: enabling device (0000 -> 0002)
[  160.015500] Device Type: Soft IP
[  160.015501] IP Type: EQDMA Soft IP
[  160.015501] Vivado Release: vivado 2020.2
[  160.015509] onic:qdma_device_attributes_get: qdma04000-p0000:04:00.0: num_pfs:1, num_qs:512, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[  160.015510] onic:qdma_device_open: Vivado version = vivado 2020.2
[  160.015512] qdma_dev_entry_create: Created the dev entry successfully
[  160.022109] onic:xdev_identify_bars: AXI Master Lite BAR 2.
[  160.022112] onic:qdma_device_open: 0000:04:00.0, 04000, pdev 0x000000001914674f, xdev 0x00000000815ae9a2, ch 1, q 1024, vf 0.
[  160.141314] onic 0000:04:00.0 enp4s0: renamed from onic4s0f0
[  160.205177] real_num_tx_queues: 68 real_num_rx_queues 68
[  160.205194] onic 0000:04:00.0: onic_cdev_ptr->name = reconic-mm
[  160.205195] onic 0000:04:00.0: successfully cdev_add a character device, reconic-mm, to the system
[  160.205351] onic 0000:04:00.0: successffully device_create a character device, reconic-mm, and register it
[  160.243862] onic 0000:04:00.0 enp4s0: onic_open: device open done
[  161.149484] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
```
After run rdma read test on the client:
```
root@niexun-HP-288-Pro-G6-Microtower-PC:/home/niexun/RecoNIC/examples/rdma_test# sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./read -r 192.100.51.1 -i 192.100.52.1 -p /sys/bus/pci/devices/0000:01:00.0/resource2 -z 128 -l dev_mem -d /dev/reconic-mm -c -u 22222 -t 11111 --dst_qp 2 -g 2>&1 | tee client_debug.log
src_ip_str = 192.100.51.1
dst_ip_str = 192.100.52.1
Info: PCIe resource file: /sys/bus/pci/devices/0000:01:00.0/resource2
Info: QP allocated at: dev_mem
Info: Device - /dev/reconic-mm
Info: src_ip = 192.100.51.1
Info: Found network interface: enp1s0
[... cut ...]
/home/antl/RecoNIC/lib/rdma_api.c:1087:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f3080c59000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/home/antl/RecoNIC/lib/rdma_api.c:1095:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f3080c59000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
[1]    20813 segmentation fault  sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./read -r 192.100.52.1 -i  -p  -z   | 
       20814 done                tee server_debug.log
```

Really appreciate for your guidance！
Xun Nie

zhguanw-amd, 2023-12-11:
@niexun725 
Hi Xun Nie,

The dmesg output seems okay.

> Besides, After use ila, for the box_250mhz module of host, the"a_axis_adap_rx_250mhz*"interface, it indeed has some input packets. However, "m_axis_user2rdma_roce_from_cmac_rx_* " interface nerver [...]

You didn't see the packets via ILA doesn't mean the remote peer (the server node) didn't receive it. This is because your ILA just missed the packets somehow when sampling. The issue should not be the packet classification. The reason is that your server node's [RN_RDMA_GCSR_ERRBUFWPTR      = 0x6006c] = 0x2, which means the server have received two RDMA packets. (It should be '1' if you run the read test only once after fresh-program. I guess probably you ran it twice somehow. Anyway, this is not important). At the server side, what you can do is either to print first entry of the error buffer at the host memory or use ILA to check this signal "axi_rdma_send_write_payload_wdata". The first 32-bit is the error syndrome, which you can find the definition in Table 2-3 from ERNIC PG332 v3.1. You should check that first to understand why your incoming RDMA packet has the issue.

The reason why you have "[RN_RDMA_GCSR_IPKTERRQWPTR    = 0x60094] = 0x1" at your client side should be that your server replies back nak read response signal.

Thanks,
Guanwen

niexun725, 2023-12-27:
@zhguanw-amd 
> The reason why you have "[RN_RDMA_GCSR_IPKTERRQWPTR = 0x60094] = 0x1" at your client side should be that your server replies back nak read response signal.
Dear Zhguanw,
Finally, I successfully ported the project to the VCU128 FPGA, thank you very much for your guidance. 
 
The problem lies in the failure to obtain the destination MAC address when running test samples such as rdma _ write, rdma _ read, etc. 
 
I run the test sample according to the steps and check the captured RDMA frames. I find that all bits of the destination MAC address are zeros, and  ernic will not work properly if this value is wrong. 
 
I found that all these tests failed to read the destination mac address correctly. However, I used ' arp-a ' to view the ARP table of the linux system, I can indeed find the destination MAC address of the NIC, so I manually configured the MAC address. After that, these tests were successfully performed.

I think there are some problems with the MAC address obtain program in write.c and read.c, and I will further check later. Besides, I will try to use HBM to replace DDR4 in RecoNIC.

Thanks,
Xun Nie

zhguanw-amd, 2023-12-28:
Hi Xun Nie,

Great to hear that it finally works at your end.

> I think there are some problems with the MAC address obtain program in write.c and read.c, and I will further check later.

I have no idea. We haven't encountered this issue and other groups using it also haven't reported any similar issues.

> Besides, I will try to use HBM to replace DDR4 in RecoNIC.

That would be nice. 

Would you mind to push your changes of VCU128 porting to the RecoNIC repository when it's ready?

Thanks,
Guanwen

niexun725, 2023-12-28:
Dear Zhguanw,
> Would you mind to push your changes of VCU128 porting to the RecoNIC repository when it's ready?

Sure, thank you for your invitation，I 'd be very happy if I could get involved.
Thanks,
Xun Nie

niexun725, 2023-12-28:
@zhguanw-amd 
> Would you mind to push your changes of VCU128 porting to the RecoNIC repository when it's ready?

Dear Zhguanw,

Thank you for your invitation，I 'd be very happy if I could get involved. I will push  my changes of VCU128  porting to the RecoNIC repository  when it's ready.

Thanks,
Xun Nie

zhguanw-amd, 2024-01-02:
@niexun725 

Great! Thanks Xun Nie! I'm going to close this issue, as you have solved it. Feel free to re-open it if you think it's necessary.

Thanks,
Guanwen

closed 2024-01-02 by zhguanw-amd