Xilinx/RecoNIC #12: Intermittent failure : CQHEADi:0x0 and SQPIi:0x1 are different on client for rdma write example

wcferry, 2024-01-24:
Greetings,

We are running vanilla Reconic per all specs on the Readme, U250 boards, correct Ubuntu ver/kernel, etc. via direct fiber between cards.
Though the server program receives the data (client's RDMA engine actually wrote data to remote) but updating to its completion queue corresponding to WQE on client side is inconsistent.

Clearing the error and continuing restores functionality but we are wondering if this is expected behavior.

Attached please find the error occur in logpc18, line 195 (pc 18 is the client)
[dmesg_pc18.log](https://github.com/Xilinx/RecoNIC/files/14031539/dmesg_pc18.log)
[kern_pc18.log](https://github.com/Xilinx/RecoNIC/files/14031540/kern_pc18.log)
[logpc18.log](https://github.com/Xilinx/RecoNIC/files/14031541/logpc18.log)

Any insight of this behavior is appreciated.
Thanks
Will

adity14, 2024-01-24:
Hello,

Could you provide information about the commit used to generate the Bitstream? Additionally, if feasible, could you examine the axi_rdma_send_write_payload_wdata using ILA? In the log, there's an observation of the error_buffer pointer ([RN_RDMA_GCSR_ERRBUFWPTR = 0x6006c] = 0xf7) having a non-zero value, indicating that the RDMA engine received error packets.

wcferry, 2024-01-24:
Certainly, and thanks for your help!
The commit # is 5e0c22d3178c664866624b5b0fefa45a39637419, from 2023-12-27 06:08:47
I am adding axi_rdma_send_write_payload_wdata (+wstrb, wvalid) to ILA to inspect per your suggestion and will report results once the build finishes
Cheers,
Will

wcferry, 2024-01-25:
Is there a particular packet content you'd like to trigger off of on the axi_rdma_send_write_payload_wdata bus?

zhguanw-amd, 2024-01-25:
Hi Will,

> Certainly, and thanks for your help! The commit # is [5e0c22d](https://github.com/Xilinx/RecoNIC/commit/5e0c22d3178c664866624b5b0fefa45a39637419), from 2023-12-27 06:08:47 I am adding axi_rdma_send_write_payload_wdata (+wstrb, wvalid) to ILA to inspect per your suggestion Cheers, Will

Could you switch to the latest commit and do the testing again? There are two fixes in the newest commit.

For the trigger, you can detect (arready & arvalid ) of axi_rdma_send_write_payload_* bus.

Thanks,
Guanwen

wcferry, 2024-02-14:
Hi Guanwen,
Sorry for the delay in response!

With checkout of the latest fpga and driver source (commit #ebf6a52) the error above still occurs, at iteration ~900 of send_recv test.

I added the signals you specified to ILA (tapped in the open_nic_shell.sv module, triggering on axis_aclk) but am not seeing the signal condition (arready & arvalid ) trigger.  ILA instantiation below.

I also triggered off of payload_wdata != 0, and did observe transitions on that bus, but not accompanied by payload arready or payload_arvalid.

Could you provide more insight on specifically what to trigger off of to capture when error actually occurs?  Also why the arrready & arvalid not asserting. 

Thanks Kindly,
Will

[
  wf_rdma_dbg_ila wf_ila_inst0 (
	.clk(axis_aclk), // input wire clk
	.probe0(axi_rdma_send_write_payload_wdata), // input wire [511:0]  probe0  
	.probe1(axi_rdma_send_write_payload_arready), // input wire [0:0]  probe1 
	.probe2(axi_rdma_send_write_payload_arvalid) // input wire [0:0]  probe2
);
]

The test we are running is send_recv on loop, with syntax below. 
[image]

ILA Capture:
[image]

amd-akolekar, 2024-02-19:
Hello Will,

Sorry for the confusion. To clarify, whenever the RDMA engine receives an error packet, it writes it to the error buffer along with the error syndrome. This is why you can monitor signals axi_rdma_send_write_payload_wdata, axi_rdma_send_write_payload_awvalid, and axi_rdma_send_write_payload_awready through ILA on the client side to get that error syndrome value. For more information refer to page 16 of [ERNIC product guide](https://docs.xilinx.com/v/u/3.0-English/pg332-ernic). Additionally, in the logs of client and server, please check the values of the RN_RDMA_GCSR_ERRBUFWPTR and RN_RDMA_GCSR_IPKTERRQWPTR registers. If these registers have a non-zero value, it indicates that an error packet has been received by the RDMA engine. 
Also, It would be helpful if you could provide logs from both the client and server sides.

Thanks,
Aditya

adity14, 2024-02-19:
Hello Will,

Sorry for the confusion. To clarify, whenever the RDMA engine receives an error packet, it writes it to the error buffer along with the error syndrome. This is why you can monitor signals axi_rdma_send_write_payload_wdata, axi_rdma_send_write_payload_awvalid, and axi_rdma_send_write_payload_awready through ILA on the client side to get that error syndrome value. For more information refer to page 16 of [ERNIC product guide](https://docs.xilinx.com/v/u/3.0-English/pg332-ernic). Additionally, in the logs of client and server, please check the values of the RN_RDMA_GCSR_ERRBUFWPTR and RN_RDMA_GCSR_IPKTERRQWPTR registers. If these registers have a non-zero value, it indicates that an error packet has been received by the RDMA engine.
Also, It would be helpful if you could provide logs from both the client and server sides.

Thanks,
Aditya

wcferry, 2024-02-26:
On further investigation, it seems that the updated checkout to #ebf6a52 I reported in my earlier post was run with only the fpga bitfile updated, but not the latest driver code rebuilt.

With the latest bitfile AND driver rebuilt to #ebf6a52 we now see a repeatable failure on first run of examples/send_recv (isolated to the root cause being the driver / rdma_send test app code update) introduced between and #5e0c22d #ebf6a52.

The commands run and logs are attached below, on server polling never completes and on client we see QP fatal error.

Your help appreciated.
Will

[image]

[wf_24_02_26_latest_send_recv_log_error_pc7.txt](https://github.com/Xilinx/RecoNIC/files/14411122/wf_24_02_26_latest_send_recv_log_error_pc7.txt)
[wf_24_02_26_latest_send_recv_log_error_PC18.txt](https://github.com/Xilinx/RecoNIC/files/14411124/wf_24_02_26_latest_send_recv_log_error_PC18.txt)

adity14, 2024-02-28:
Can you see changes from commit [#60fd0](https://github.com/Xilinx/RecoNIC/commit/60fd030930f3a760c94243a778484d8fa0912b40) in your code?

zhguanw-amd, 2024-03-04:
@wcferry 
Hi Will,

Can you also try with the read/write commands shown in the [README.md](https://github.com/Xilinx/RecoNIC?tab=readme-ov-file#rdma-test-cases)? Any issues at your end?

wcferry, 2024-03-08:
Tested RDMA read command, with same bitfile (latest) and sw driver (latest), we see failure to complete, logs attached.

Would it be possible to reproduce on your end with the latest sw build?

[24_03_08_latestcommit_rdma_read_pc18_server.txt](https://github.com/Xilinx/RecoNIC/files/14543111/24_03_08_latestcommit_rdma_read_pc18_server.txt)
[24_03_08_latestCommit_rdmaRead_pc7_Client.txt](https://github.com/Xilinx/RecoNIC/files/14543112/24_03_08_latestCommit_rdmaRead_pc7_Client.txt)

zhguanw-amd, 2024-03-09:
@wcferry 
Hi Will,

From the log, I can see that you get stuck with the TCP handshanking stage. Something wrong with your setup. I suggest you to start a fresh clone and build from scratch. The release on the repo has been tested by us and many other internal/external groups as well. It should be simple to set up on U250.

wcferry, 2024-03-15:
Working on this
Appreciating all your help

in the meantime, could you tell me, does the reconic project support using 2 MACS connected to the Ernic core in the current state of the project, (via num_cmac_port parameter in build.tcl), or is this just a feature in the underlying open_nic project not supported in reconic?

Thanks
Will

zhguanw-amd, 2024-03-16:
Hi Will,

The current RecoNIC doesn't use the two MACs configuration in the OpenNIC. To support this, either to have two RDMA datapath or share the two MACs with single RDMA datapath. This task will require a certain amount of RTL work.

zhguanw-amd, 2024-04-22:
No follow-up discussion for a while. I'm going to close this thread. Feel free to re-open it, if you have further questions.

closed 2024-04-22 by zhguanw-amd