Xilinx/RecoNIC #13: Can we modify it to support U200 and U50 ? How to do it ?

torukskywalker, 2024-01-26:
We only have U200 and U50, how to modify to support these boards ?

zhguanw-amd, 2024-01-26:
U200 should be easy by adding -board au200 when calling vivado in .RecoNIC/scripts/Makefile: vivado -mode batch -source build.tcl -tclargs -board=$(BOARD) -max_pkt_len 9600 -impl 1 -overwrite 1 -board_repo $(BOARD_REPO), where BOARD=au200.

U200/U250 has DDR, while U50 has HBM. The memory controller should be different. Thus, you need to port HBM's memory controller for it.

zhguanw-amd, 2024-01-30:
Feel free to re-open the issue if needed.

closed 2024-01-30 by zhguanw-amd

function47, 2024-02-21:
Hi, Guanwen, 
For the HBM board (U280), can we start by building the shell without using the HBM?

zhguanw-amd, 2024-02-22:
Hi Zhaoyang,

To build U280 without HBM, you can simply remove ddr controller interface in open_nic_shell.sv, update ./scripts/Makefile for adding new board support and remove "dev_mem_ddr4_controller" in ./base_nics/open-nic-shell/src/utility/vivado_ip/vivado_ip.tcl. In this case, you won't be able to use any features related to the device memory.

You can start a new thread for this if you have further questions, as it's a new topic.

Thanks,
Henry