Xilinx/RecoNIC #15: Support it on Alveo U280

function47, 2024-02-22:
Create a new issue from: https://github.com/Xilinx/RecoNIC/issues/13#issuecomment-1958287545

Q: For the HBM board (U280), can we start by building the shell without using the HBM?

A: To build U280 without HBM, you can simply remove ddr controller interface in open_nic_shell.sv, update ./scripts/Makefile for adding new board support and remove "dev_mem_ddr4_controller" in ./base_nics/open-nic-shell/src/utility/vivado_ip/vivado_ip.tcl. In this case, you won't be able to use any features related to the device memory.

function47, 2024-02-22:
Hi, Henry,

Since the U280 also have 8GB DDR memory on board, Is it also possible to preserve the feature of using DDR memory? We can still use the DDR memory controller to access the U280 memory, can we?

zhguanw-amd, 2024-02-22:
Yes, it's possible. But you might need to update the constraint file and dev_mem_ddr4_controller.tcl for U280, if its DDR configuration is different from U250.

function47, 2024-02-22:
Perfect! That is what I guess! 

Thanks!

closed 2024-02-23 by zhguanw-amd