Xilinx/RecoNIC #16: synth_1 failed with named port connection does not exist for instance 'qdma_inst' of module 'qdma_no_sriov'

qianyich, 2024-02-24:
Greetings,

I am using Vivado 2021.2 and trying to port the platform to Alveo U280. When building the hardware:

```
[Sat Feb 24 16:36:44 2024] Launched box_322mhz_axi_crossbar_synth_1, synth_1...
Run output will be captured here:
box_322mhz_axi_crossbar_synth_1: /home/qianyich/RecoNIC/base_nics/open-nic-shell/build/au280/open_nic_shell/open_nic_shell.runs/box_322mhz_axi_crossbar_synth_1/runme.log
synth_1: /home/qianyich/RecoNIC/base_nics/open-nic-shell/build/au280/open_nic_shell/open_nic_shell.runs/synth_1/runme.log
[Sat Feb 24 16:36:44 2024] Launched impl_1...
Run output will be captured here: /home/qianyich/RecoNIC/base_nics/open-nic-shell/build/au280/open_nic_shell/open_nic_shell.runs/impl_1/runme.log
[Sat Feb 24 16:36:44 2024] Waiting for impl_1 to finish...
[Sat Feb 24 16:39:06 2024] impl_1 finished
WARNING: [Vivado 12-8222] Failed run(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:02:17 ; elapsed = 00:02:21 . Memory (MB): peak = 4868.984 ; gain = 0.000 ; free physical = 41299 ; free virtual = 88058
# if {$post_impl} {
#     _do_post_impl $top_build_dir $top impl_1 $zynq_family
# }
INFO: [Common 17-206] Exiting Vivado at Sat Feb 24 16:39:06 2024...
```

the following is part of the synth_1 runme.log

```
ERROR: [Synth 8-448] named port connection 'st_rx_msg_rdy' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:601]                                                                                                                                                                                               ERROR: [Synth 8-448] named port connection 'st_rx_msg_valid' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:602]                                                                                                                                                                                             ERROR: [Synth 8-448] named port connection 'st_rx_msg_last' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:603]                                                                                                                                                                                              ERROR: [Synth 8-448] named port connection 'st_rx_msg_data' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:604]                                                                                                                                                                                              ERROR: [Synth 8-448] named port connection 'csr_prog_done' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:748]                                                                                                                                                                                               ERROR: [Synth 8-448] named port connection 's_axil_csr_awaddr' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:749]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_awprot' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:750]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_awvalid' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:751]                                                                                                                                                                                          ERROR: [Synth 8-448] named port connection 's_axil_csr_awready' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:752]                                                                                                                                                                                          ERROR: [Synth 8-448] named port connection 's_axil_csr_wdata' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:753]                                                                                                                                                                                            ERROR: [Synth 8-448] named port connection 's_axil_csr_wstrb' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:754]                                                                                                                                                                                            ERROR: [Synth 8-448] named port connection 's_axil_csr_wvalid' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:755]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_wready' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:756]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_bvalid' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:757]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_bresp' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:758]                                                                                                                                                                                            ERROR: [Synth 8-448] named port connection 's_axil_csr_bready' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:759]                                                                                                                                                                                           ERROR: [Synth 8-448] named port connection 's_axil_csr_araddr' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:760]
ERROR: [Synth 8-448] named port connection 's_axil_csr_arprot' does not exist for instance 'qdma_inst' of module 'qdma_no_sriov' [/home/qianyich/RecoNIC/base_nics/open-nic-shell/src/qdma_subsystem/qdma_subsystem_qdma_wrapper.v:761]
[... cut ...]                                                                                                                                                          ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details                                                                                                      INFO: [Common 17-206] Exiting Vivado at Sat Feb 24 16:39:00 2024...
```

These are ports of QDMA bridge CSR interface and slave interface.

qianyich, 2024-02-25:
remove those connections seems to synthesize successfully.

zhguanw-amd, 2024-02-26:
Hi Qianyi,

You are using the default _qdma_no_sriov_au280.tcl_ located at _base_nics/open-nic-shell/src/qdma_subsystem/vivado_ip_, which is not right. The default one doesn't support QDMA AXI-MM channel and PCIe slave bridge, that's why you can't see those ports. It's not the right way, if you simply remove those ports. You can copy _qdma_no_sriov_au250.tcl_, rename it as au280 and change _CONFIG.xlnx_ref_board {AU250}_ to AU280 in this file. It should solve this issue.

Thanks,
Guanwen

qianyich, 2024-02-26:
Thank you Guanwen. That helps solve the issue.

but I am still having a small setup time violation which is about 0.161. I am not sure if it can be eliminated through a more aggressive implementation strategy, but this might not work in the future as users add more logic.

Regarding what I changed to port the design to U280, I simply changed CONFIG.C0_CLOCK_BOARD_INTERFACE {default_300mhz_clk0} to {sysclk0} and CONFIG.C0.DDR4_Specify_MandD to true. This allows to set the DDR reference input clock speed to 300MHz on U280. The rest are the same since the floorplanning of U280 (the location of DDR and QSFP) is quite similar to what U250 has.

zhguanw-amd, 2024-02-27:
For the timing violation, probably you can try with several implementation strategies. It should give you a version without timing violation, as your timing violation is around ~0.1. 

> but this might not work in the future as users add more logic.
Actually, this is not true. The timing issue at the moment is because of the locations of harden QDMA and CMAC IPs. It's not caused by the resource congestion. Sometimes, if you add more logic, it becomes easier to meet the timing.

qianyich, 2024-02-27:
Thank you Guanwen. That makes sense. I went ahead using Vivado IDR flow to implement the design, and now there is no timing violation. However, there is still no MAC address assigned to the device. We can use the other issue post. This can be closed.

closed 2024-02-27 by qianyich