Xilinx/RecoNIC #19: Error: failed to lock page in memory

qianyich, 2024-03-21:
When I was trying to run rdma_test read and write, I have the following error.

```
qianyich@pc164:~/RecoNIC/examples/rdma_test$ sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./write -r 192.100.51.1 -i 192.100.52.1 -p /sys/bus/pci/devices/0000\:3b\:00.0/resource2 -z 128 -l host_mem -d /dev/reconi
c-mm -c -u 22222 -t 11111 --dst_qp 2 -g 2>&1 | tee client_debug.log
src_ip_str = 192.100.51.1
dst_ip_str = 192.100.52.1
Info: mac_addr_t = 00:0a:35:fd:c0:a8
Info: PCIe resource file: /sys/bus/pci/devices/0000:3b:00.0/resource2
Info: QP allocated at: host_mem
Info: Device - /dev/reconic-mm
Info: src_ip = 192.100.51.1
Info: Found network interface: enp59s0
Info: mac_addr_t = 00:0a:35:c0:8c:15
Info: Creating rn_dev
/users/qianyich/RecoNIC/lib/reconic.c:296:create_rn_dev(): Info: scr(=4)) file open successfully
create_rn_dev - testing2
Error: failed to lock page in memory
```

I found this in lib/reconic.c:323. Is this due to insufficient huge page? Should I enable and configure the number of huge pages in Linux? What exactly should I do in this case?

zhguanw-amd, 2024-03-22:
Hi Yicheng,

Could you double check whether your TCP is working fine?

To set up RDMA, we need to first have the QP connection, which requires two peers to exchange some sort of information. The information includes such as r_key, remote addresses and psn. To exchange the information, it can be done via connection management with QP0 or TCP. We are using TCP for exchanging such information.

qianyich, 2024-03-22:
Hi Guanwen,

I used tcping between my two servers, and it just works fine.

```
8 probes transmitted on port 22 | 8 received, 0.00% packet loss
successful probes:   8
unsuccessful probes: 0
last successful probe:   2024-03-21 22:26:29
last unsuccessful probe: Never failed
total uptime:   8 seconds
total downtime: 0.0 seconds
longest consecutive uptime:   7 seconds from 2024-03-21 22:26:22 to 2024-03-21 22:26:30
retried to resolve hostname 0 times
rtt min/avg/max: 0.223/0.266/0.324 ms
--------------------------------------
TCPing started at: 2024-03-21 22:26:22
TCPing ended at:   2024-03-21 22:26:30
duration (HH:MM:SS): 00:00:08
```

qianyich, 2024-03-22:
Oh I guess you actually mean check TCP connection between 192.100.52.1 and 192.100.51.1. So I should use tcping 192.100.52.1 in this case. Let me do this and will get back to you.

qianyich, 2024-03-22:
I saw others have the same issue. https://github.com/Xilinx/RecoNIC/issues/12#issuecomment-1986413626 Do you have any suggestions on solving this TCP handshaking issue?

zhguanw-amd, 2024-03-23:
Hi Yicheng,

Could you comment out all the code after exchanging the remote offset and see what's the issue? Actually, at this point, your program hasn't reached to the actual RDMA part, only a few RDMA register confgurations.

qianyich, 2024-03-23:
From which line to which line? Do you mean comment out all actual rdma part(wqe,etc) or this problematic part? Please bear with me, I am still trying to understand what those codes are trying to do.

When I did tcping, should I use 11111 as the port number? I saw tcp port is 11111 in the command arguments, but I did not verify if that works or not.

adity14, 2024-03-25:
Hello,

Can you try with a larger payload size like 4096 or above by changing argument -z from 128 to 4096 in your command

Aditya.

qianyich, 2024-03-25:
Hi Aditya,

I did reprogram them every time before running the test. I will change the payload size and try today, but it seems like the problem is the TCP handshaking.

Yicheng

qianyich, 2024-03-26:
**I think the problem is with the dma engine. I ran the dma test but it failed on reading from the device memory.**

```
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
size=65536000 Average BW = 10.629003 GB/sec, average latency = 6165.771279 us
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
/dev/reconic-mm, read off 0x0 + 0x3e80000 failed -1.
read file: Invalid argument
```

Then I see that the read_to_buffer always fails in read, write, systolic array test (rc == -5 returned from read_to_buffer). For the systolic array, I have the following logs:
[client_debug.log](https://github.com/Xilinx/RecoNIC/files/14763987/client_debug.log)
[server_debug.log](https://github.com/Xilinx/RecoNIC/files/14763988/server_debug.log)

client:
sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./network_systolic_mm -d /dev/reconic-mm -p /sys/bus/pci/devices/0000\:3b\:00.0/resource2 -r 192.100.52.1 -i 192.100.51.1 -u 22222 -t 11111 --dst_qp 2 -c 2>&1 | tee client_debug.log

server:
sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./network_systolic_mm -d /dev/reconic-mm -p /sys/bus/pci/devices/0000\:3b\:00.0/resource2 -r 192.100.51.1 -i 192.100.52.1 -u 22222 -t 11111 --dst_qp 2 -s 2>&1 | tee server_debug.log

zhguanw-amd, 2024-03-27:
@qianyich 
Hi Yicheng,

Check your log below:

`/users/qianyich/RecoNIC/lib/rdma_api.c:550:allocate_rdma_qp(): DEBUG: qp->sq->dma_addr = 0x2f53422000, sq_addr_msb = 0xf, sq_addr_lsb = 0x53422000`

In RecoNIC's QDMA setup, we use 1TB as the AXI BAR0 (which uses PCIe slave bridge to read/write host data) and it gets further splited into 8 windows equally. This means that the BDF mask is for 128GB. I guess you're using 256GB system, right? Your dma_address is not matched with {sq_addr_msb, sq_addr_lsb}, which means that the hardware can't read any WQE from the SQ. That's why you can make it work.

To enable 256GB mask, you need to configure the BDF table's 0x2430 register with 0xC400_0000. You can check the information [here](https://docs.amd.com/r/4.0-English/pg302-qdma/BDF-Table).

Regarding the DMA engine issue, probably your hardware configuration or connection has some problems when porting to U280. We never encounter this issue. What you could do is to check the built-in example along with the QDMA 4.0 on U280, see any special configuration. To build the example, please check [this](https://support.xilinx.com/s/article/72352?language=en_US).

qianyich, 2024-03-27:
Hi Guanwen,

I took a look at this https://support.xilinx.com/s/article/Demystifying-BDF-Table-programming-for-Slave-Bridge-Address-Translations-for-AXI-address?language=en_US. And it does help me understand this better. I understand the address above goes over the range of 128 GB, and the S_AXI_BRIDGE is set to be 1TB, but I am not sure if I am using 256 GB, how can I check this and how can I write 0x2430 with C4000000 (I do understand it is from 256 * 1024 * 1024 / 4)?

Will check the configuration of the example QDMA project later and keep you posted.

zhguanw-amd, 2024-03-27:
@qianyich I mean your server probably has 256GB host memory, that's why you will have addresses above 128GB range when you allocate memory. 

To write 0x2430, please check [this line](https://github.com/Xilinx/RecoNIC/blob/main/lib/reconic.c#L184). "AXIB_BDF_MAP_CONTROL_ADDR" is 0x2430 register.

qianyich, 2024-03-28:
@zhguanw-amd 
Hi Guanwen,

I generated the example QDMA design, and tested it with dma_ip_drivers. My script tests the functionality of QDMA.

```
qdma3b000       0000:3b:00.0    max QP: 32, 0~31
=============Hardware Version============

RTL Version         : RTL Base
Vivado ReleaseID    : vivado 2020.2
QDMA Device Type    : Soft IP
QDMA IP Type    : EQDMA4.0 Soft IP
============Software Version============

qdma driver version : 2022.1.5.5.

=============Hardware Capabilities============

Number of PFs supported                : 1
Total number of queues supported       : 512
MM channels                            : 1
FLR Present                            : no
ST enabled                             : yes
MM enabled                             : yes
Mailbox enabled                        : no
MM completion enabled                  : no
Debug Mode enabled                     : no
Desc Engine Mode                       : Inernal only mode
qdma3b000:statistics
Total MM H2C packets processed = 17000
Total MM C2H packets processed = 17
Total ST H2C packets processed = 0
Total ST C2H packets processed = 0
Min Ping Pong Latency = 0
Max Ping Pong Latency = 0
Avg Ping Pong Latency = 0
-e
List Xilinx PCIe QDMA queues.
Zero Qs-e
Test Xilinx PCIe QDMA h2c channel.
qdma3b000-MM-0 H2C added.
Added 1 Queues.
/dev/qdma3b000-MM-0
dma-ctl: Info: Default ring size set to 2048
1 Queues started, idx 0 ~ 0.
size=4096 Average BW = 368.020416 MB/sec
size=65536 Average BW = 3.080631 GB/sec
Stopped Queues 0 -> 0.
Deleted Queues 0 -> 0.
qdma3b000:statistics
Total MM H2C packets processed = 34000
Total MM C2H packets processed = 17
Total ST H2C packets processed = 0
Total ST C2H packets processed = 0
Min Ping Pong Latency = 0
Max Ping Pong Latency = 0
Avg Ping Pong Latency = 0
-e
Test Xilinx PCIe QDMA c2h channel.
qdma3b000-MM-0 C2H added.
Added 1 Queues.
/dev/qdma3b000-MM-0
dma-ctl: Info: Default ring size set to 2048
1 Queues started, idx 0 ~ 0.
size=4096 Average BW = 336.469760 MB/sec
size=65536 Average BW = 3.472523 GB/sec
Stopped Queues 0 -> 0.
Deleted Queues 0 -> 0.
qdma3b000:statistics
Total MM H2C packets processed = 34000
Total MM C2H packets processed = 17017
Total ST H2C packets processed = 0
Total ST C2H packets processed = 0
Min Ping Pong Latency = 0
Max Ping Pong Latency = 0
Avg Ping Pong Latency = 0
-e
Test done.
```
Here are some differences I observed between this example QDMA project and the one in open-nic-shell:
1. Descriptor bypass is disabled in the built-in QDMA example, but is on in open-nic-shell
2. The BAR for AXI lite master in open-nic-shell is 4MB and has a value of FFFFFFFFFFC00004. It is FFFFFFFFFFFFF004 in the example with a size of 4KB. This shouldn't cause any issues.
[image]
3. QDMA is using advanced mode in open-nic-shell which is reasonable because we need AXI-lite CSR slave interface.

Should I disable descriptor bypass? I haven't tried running with changing BDF table entry. Maybe that will fix the issue.

qianyich, 2024-03-28:
Is it possible because the evaluate ERNIC IP license cracks the integrity of the whole bitstream after certain hours? And now the DMA is not working?

zhguanw-amd, 2024-03-28:
@qianyich 

> Is it possible because the evaluate ERNIC IP license cracks the integrity of the whole bitstream after certain hours? And now the DMA is not working?

The two won't affect each other.

I download the source from the repo and try it today. I also have the same issue for the DMA part. It seems we push a version, which causes the DMA read-side problem. I'll revert the version and fix it next week (this week is too busy). Thanks and sorry for the trouble! The other issue mentioned above should be related to your system setup.

qianyich, 2024-03-28:
Sounds good! Thank you so much Guanwen!

qianyich, 2024-04-01:
Any progress on this?

zhguanw-amd, 2024-04-02:
@qianyich You can try the latest commit, there is a fix in the driver side.

qianyich, 2024-04-02:
I am now having a problem loading the driver. The driver got compiled, but it failed to initialize the device.

```
[ 1173.819980] pci 0000:3b:00.0: [10ee:903f] type 00 class 0x058000
[ 1173.819998] pci 0000:3b:00.0: reg 0x10: [mem 0xab400000-0xab43ffff 64bit]
[ 1173.820005] pci 0000:3b:00.0: reg 0x18: [mem 0xab000000-0xab3fffff 64bit]
[ 1173.820023] pci 0000:3b:00.0: enabling Extended Tags
[ 1173.820230] pci 0000:3b:00.0: BAR 2: assigned [mem 0xab000000-0xab3fffff 64bit]
[ 1173.820236] pci 0000:3b:00.0: BAR 0: assigned [mem 0xab400000-0xab43ffff 64bit]
[ 1173.820360] onic 0000:3b:00.0 onic59s0f0 (uninitialized): Set MAC address to 0:a:35:a:3f:f7
[ 1173.820362] onic 0000:3b:00.0: device is a master PF
[ 1173.820364] onic_set_num_queue: num_msix 8, nb_queues 7, pci_msix_user_cnt 1
[ 1173.820365] onic_pci_probe mm_queues: 4
[ 1173.820417] onic:qdma_device_open: onic, 3b:00.00, pdev 0x00000000c9cf9db1, 0x10ee:0x903f.
[ 1173.820557] qdma_is_config_bar: Invalid config bar, err:-4
[ 1173.826056] qdma_hw_access_init: config bar passed is INVALID, err:-1
[ 1173.832546] onic 0000:3b:00.0: onic_qdma_setup: qdma_device_open() failed: Error Code: -22
[ 1173.840808] onic 0000:3b:00.0: onic_pci_probe: onic_qdma_setup() failed with status -22
[ 1173.848869] onic: probe of 0000:3b:00.0 failed with error -22
```

zhguanw-amd, 2024-04-03:
Ah okay. Please re-enable the part (Line 1158 - 1166 in onic_enable_cmac() located at onic_main.c) I commented out. It should work with your setup. I guess you're having two boards connected directly.

qianyich, 2024-04-03:
hmm, the driver can be loaded, but the dma is still problematic. Both read and write failed. BTW, my two boards are not connected directly. They are connected to network switch.

```
qianyich@pc164:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
/dev/reconic-mm, W off 0x0, 0x3e80000 failed -1.
write file: Input/output error
qianyich@pc164:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
/dev/reconic-mm, read off 0x0 + 0x3e80000 failed -1.
read file: Input/output error
```

zhguanw-amd, 2024-04-03:
Hi Yicheng,

Can you put some pr_info in the onic_main.c and get (q_no, qhndl, xpriv->base_rx_q_handle, queue_id, q_handle, xpriv->pinfo->active_rx_queues, xpriv->pinfo->active_tx_queues) in onic_rx_pkt_process(), onic_rx_poll(), onic_isr_rx_tophalf(), onic_qdma_rx_queue_add(), onic_isr_tx_tophalf() and onic_qdma_tx_queue_setup()? And please share the dmesg log with me, so that I can understand what's going on at your end.

For the pr_info, do remember to put "__func__", so that I can know which function.

Thanks.

zhguanw-amd, 2024-04-03:
Regarding your rdma question, you can check ERNIC 3.1 document's Chapter 2 Register Space, there are some status registers available, which provide internal information of the RDMA engine. We are planning to upgrade it to version 4.0.

Did you solve the address issue I mentioned before by configuring more BDF table and removing address masking? If you haven't done that, the register information has no meaning, as the RDMA engine is not working properly. If you don't know how to do it, please wait for one or two weeks and I'll update it to include larger host memory up to 1TB.

qianyich, 2024-04-03:
The BDF table entry is now correct and the memory address space configuration is all good. However, I think it would be better to make it support a larger memory by default.

I will add those pr_info by tomorrow. Just noticed that some variables are not valid in the scope of some functions, and I will just ignore those.

zhguanw-amd, 2024-04-03:
Probably you can send me your modifications to support u280 (henry.zhong AT amd.com). I can take a look and check it next week.

qianyich, 2024-04-03:
I am not sure what is going on, but after I inserted pr_info lines, reboot the machine, reload the driver. The dma starts to work

```
qianyich@pc164:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
size=65536000 Average BW = 4.948909 GB/sec, average latency = 13242.515735 us
qianyich@pc164:~/RecoNIC/examples/dma_test$ sudo  ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
size=65536000 Average BW = 5.257542 GB/sec, average latency = 12465.140782 us
```

zhguanw-amd, 2024-04-03:
Okay, probably your machine/FPGA was in wrong status when you loaded the driver. Can you start a fresh one by re-programing the FPGA board and loading the driver? See whether it works.

Regarding the BDF table, I realize that QDMA set the 8 windows equally, which means that each window will have up to 128GB mapping. The previous "0xC400_0000" mentioned is wrong. I'll update the correct one next week, as my bandwidth this week is full now.

qianyich, 2024-04-03:
The send_recv test failed as well. The receiver side just hang there. The sender side just failed to perform send operation.

Receiver Side:

```
sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./send_recv -r 192.100.51.1 -i 192.100.52.1 -p /sys/bus/pci/devices/0000\:3b\:00.0/resource2 -z 128 -l host_mem -d /dev/re
conic-mm -c -u 22222 --dst_qp 2 -g 2>&1 | tee client_debug.log
src_ip_str = 192.100.51.1
dst_ip_str = 192.100.52.1
Info: mac_addr_t = 00:0a:35:5b:e2:96
Info: PCIe resource file: /sys/bus/pci/devices/0000:3b:00.0/resource2
Info: QP allocated at: host_mem
Info: Device - /dev/reconic-mm
Info: src_ip = 192.100.51.1
Info: Found network interface: enp59s0
Info: mac_addr_t = 00:0a:35:58:dd:fe
Info: CREATE RecoNIC DEVICE
/users/qianyich/RecoNIC/lib/reconic.c:297:create_rn_dev(): Info: scr(=4)) file open successfully
create_rn_dev - testing2
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ed000
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ed000000
Info: pre-allocated hugepage buffer vir addr = 0x7fb732400000, physical addr = 0x16ed000000
Info: Configuring QDMA AXI bridge BDF
/users/qianyich/RecoNIC/lib/reconic.c:195:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16420, bdf_addr_low=0x0
/users/qianyich/RecoNIC/lib/reconic.c:196:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16424, bdf_addr_high=0x0
/users/qianyich/RecoNIC/lib/reconic.c:197:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16430, bdf_win_config=0xc4000000
Info: CREATE RDMA DEVICE
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ed000
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ed000000
/users/qianyich/RecoNIC/lib/reconic.c:228:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7fb732400000, physical addr = 16ed000000, rn_dev->buffer_offset = 0x200000
/users/qianyich/RecoNIC/lib/reconic.c:229:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ed200
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ed200000
/users/qianyich/RecoNIC/lib/reconic.c:228:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7fb732600000, physical addr = 16ed200000, rn_dev->buffer_offset = 0x1200000
/users/qianyich/RecoNIC/lib/reconic.c:229:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ec200
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ec200000
/users/qianyich/RecoNIC/lib/reconic.c:228:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7fb733600000, physical addr = 16ec200000, rn_dev->buffer_offset = 0x1202000
/users/qianyich/RecoNIC/lib/reconic.c:229:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ec202
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ec202000
/users/qianyich/RecoNIC/lib/reconic.c:228:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7fb733602000, physical addr = 16ec202000, rn_dev->buffer_offset = 0x1212000
/users/qianyich/RecoNIC/lib/reconic.c:229:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/users/qianyich/RecoNIC/lib/reconic.c:145:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x16ec212
/users/qianyich/RecoNIC/lib/reconic.c:150:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/users/qianyich/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x16ec212000
/users/qianyich/RecoNIC/lib/reconic.c:228:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7fb733612000, physical addr = 16ec212000, rn_dev->buffer_offset = 0x1222000
/users/qianyich/RecoNIC/lib/reconic.c:229:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
Info: OPEN RDMA DEVICE
/users/qianyich/RecoNIC/lib/rdma_api.c:186:config_rdma_global_csr(): [Register] RN_RDMA_GCSR_DATBUFBA=0x600a0, value=0xed200000
/users/qianyich/RecoNIC/lib/rdma_api.c:188:config_rdma_global_csr(): [Register] RN_RDMA_GCSR_DATBUFBAMSB=0x600a4, value=0x16
[... cut ...]
Info: [RN_RDMA_QCSR_STATRESPSNi     = 0x60390] = 0xabc
Info: [RN_RDMA_QCSR_STATRQBUFCAi    = 0x60394] = 0x16000
Info: [RN_RDMA_QCSR_STATWQEi        = 0x60398] = 0x0
Info: [RN_RDMA_QCSR_STATRQPIDBi     = 0x6039c] = 0x0
Info: [RN_RDMA_QCSR_STATRQBUFCAMSBi = 0x603d8] = 0xa3500000
Info: [RN_RDMA_QCSR_SQPIi           = 0x60338] = 0x1

Error: Failed to perform an RDMA send operation!
Warning: CQHEADi and SQPIi for QP2 are mismatched

***** QP2 FATAL RECOVERY *****
TIMEOUT: CQHEADi:0x0 and SQPIi:0x1 are different
```

zhguanw-amd, 2024-04-03:
Hi Yicheng,

I'm not surprised that it fails, as the BDF table configuration is still wrong at your end. We can't simply set the window size to 256GB, the max is 128GB per window. Please wait for a few days and I'll push one next week to cover 8 windows up to 1TB system memory mapping.

qianyich, 2024-04-03:
Oh yeah, I forgot about it. That makes sense. Thank you!

qianyich, 2024-04-09:
This could be a stupid question, but I noticed that the transport layer protocol in RoCE v2 spec is Infiniband, is the Infiniband header used in ERNIC or RecoNIC? I am asking because I did not find Infiniband in the ERNIC document.

zhguanw-amd, 2024-04-10:
@qianyich Please try branch "map_large_hmem". Let me know if it works or not.

The "Infiniband" you mentioned is link layer, which compared to "ethernet". RoCEv2 means RDMA protocol on top of ethernet. The RDMA protocol here is IB transport protocol in your figure, which includes base transport header (BTH), extended transport header (RETH, AETH, ImmDt, IETH, AtomicETH, AtomicAckETH, ...), etc.

qianyich, 2024-04-10:
@zhguanw-amd I am observing some unstable behaviors. 

After the FPGAs were programmed, I ran the read, write, send_recv test immediately. They were all working fine. However, the systolic array example failed with result mismatch error. I then ran the dma_test on the client side, it showed some thing similar to the following:
```
/dev/reconic-mm, W off 0x0, 0x3e80000 failed -1.
write file: Input/output error
```

I rebooted the two servers, I ran the systolic array example, and it succeeded. But all rdma verbs test seems to be quite strange.
For example, rdma_read shows "Data read successfully" which means success but the log shows "Failed to send an RDMA read operation".

The logs for network_systolic_array after reboot are attached here:
[client_debug.log](https://github.com/Xilinx/RecoNIC/files/14934628/client_debug.log)
[server_debug.log](https://github.com/Xilinx/RecoNIC/files/14934629/server_debug.log)

The logs for rdma_read after reboot are attached here:
[client_debug.log](https://github.com/Xilinx/RecoNIC/files/14934647/client_debug.log)
[server_debug.log](https://github.com/Xilinx/RecoNIC/files/14934648/server_debug.log)

Also, I noticed that at the bottom of every log. I suspect those tests or some of the tests break the QDMA Queue Pairs somehow. It shouldn't be related to RDMA QPs though.

```
Warning: CQHEADi and SQPIi for QP2 are mismatched

***** QP2 FATAL RECOVERY *****
TIMEOUT: CQHEADi:0x0 and SQPIi:0x1 are different
```

qianyich, 2024-04-10:
I reprogrammed the devices, and started from the read test:
What I did in order is (1) dma test (2) read test (3) write test (4) send_recv test (5) systolic array test (6) dma test

(1) dma test passed with no error
(2) read test passed with logs attached here:
[client_debug_rdma_read.log](https://github.com/Xilinx/RecoNIC/files/14935569/client_debug_rdma_read.log)
[server_debug_rdma_read.log](https://github.com/Xilinx/RecoNIC/files/14935572/server_debug_rdma_read.log)

(3) write test passed with logs attached here:
[client_debug_rdma_write.log](https://github.com/Xilinx/RecoNIC/files/14935612/client_debug_rdma_write.log)
[server_debug_rdma_write.log](https://github.com/Xilinx/RecoNIC/files/14935616/server_debug_rdma_write.log)

(4) send_recv test passed with logs attached here:
[client_debug_receiver.log](https://github.com/Xilinx/RecoNIC/files/14935620/client_debug_receiver.log)
[server_debug_sender.log](https://github.com/Xilinx/RecoNIC/files/14935624/server_debug_sender.log)

(5) systolic array test **failed** with logs attached here (for this 192.100.52.1 is the client who requests matrices):
[server_debug_systolic.log](https://github.com/Xilinx/RecoNIC/files/14935642/server_debug_systolic.log)
[client_debug_systolic.log](https://github.com/Xilinx/RecoNIC/files/14935645/client_debug_systolic.log)

This failed and I observed errors not only for systolic array test but also read, write, send_recv test in dmesg as shown below:

192.100.51.1:
```
[  159.273862] IPv6: ADDRCONF(NETDEV_UP): enp59s0: link is not ready
[  159.273888] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0: link becomes ready
[  240.930586] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  275.085286] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  427.603894] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  427.603918] read[2476]: segfault at 29 ip 00007ffa0f6ed8f9 sp 00007ffc518e0d30 error 4 in libreconic.so[7ffa0f6e8000+c000]
[  632.051171] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  632.051203] write[2496]: segfault at 29 ip 00007fd4508358f9 sp 00007ffeac2ba5f0 error 4 in libreconic.so[7fd450830000+c000]
[  795.233063] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  795.233083] send_recv[2512]: segfault at 29 ip 00007f677f3f88f9 sp 00007ffc7484ec00 error 4 in libreconic.so[7f677f3f3000+c000]
[ 1106.850350] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1106.850376] network_systoli[2555]: segfault at 29 ip 00007f0dc6bde8f9 sp 00007ffd5c084bd0 error 4 in libreconic.so[7f0dc6bd9000+c000]
[ 1222.657479] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1228.868558] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
```

192.100.52.1:
```
[   75.281512] IPv6: ADDRCONF(NETDEV_UP): enp59s0: link is not ready
[   75.281520] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0: link becomes ready
[  103.193176] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  111.940145] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  695.196025] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  695.196050] read[2444]: segfault at 29 ip 00007f74f74758f9 sp 00007ffd6ae2c4d0 error 4 in libreconic.so[7f74f7470000+c000]
[  811.056709] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  811.056739] write[2523]: segfault at 29 ip 00007f282d1b18f9 sp 00007fff51091c90 error 4 in libreconic.so[7f282d1ac000+c000]
[ 1075.081479] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1075.081503] send_recv[2589]: segfault at 29 ip 00007fabcbb348f9 sp 00007ffe19910800 error 4 in libreconic.so[7fabcbb2f000+c000]
[ 1249.158570] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
```

(6) dma test again. This time, dma test passed on 192.100.51.1, but partially failed on 192.100.52.1!

192.100.51.1:
```
qianyich@pc166:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
size=65536000 Average BW = 4.875006 GB/sec, average latency = 13443.265110 us
qianyich@pc166:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
size=65536000 Average BW = 5.177333 GB/sec, average latency = 12658.255175 us
```

192.100.52.1:
```
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
size=65536000 Average BW = 4.914677 GB/sec, average latency = 13334.752992 us
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
/dev/reconic-mm, W off 0x0, 0x3e80000 failed -1.
write file: Input/output error
```

zhguanw-amd, 2024-04-11:
@qianyich I'll check it tomorrow on U250 and see any issues. Porbably it's related to QMDA's queue assignment.

qianyich, 2024-04-12:
@zhguanw-amd Could you merge "map_large_hmem" into the main branch so that I can see if those changes work fine on my side?

qianyich, 2024-04-12:
Nevermind, I think you might think of merging it with other changes later. FYI, the latest commit in the main branch allows me to install the driver and ping each other. I did not test those APIs as they are not fixed yet.

qianyich, 2024-04-12:
Also, I am interested in adding QSPI flash into the design so that I can program the board over PCIe. https://github.com/Xilinx/open-nic-shell/blob/add_qspi_and_cms/script/generate_qspi_block_design.tcl might be the good starting point.

zhguanw-amd, 2024-04-15:
@qianyich please try the new version in the main branch. It should fix your issues in this version.

>Also, I am interested in adding QSPI flash into the design so that I can program the board over PCIe. https://github.com/Xilinx/open-nic-shell/blob/add_qspi_and_cms/script/generate_qspi_block_design.tcl might be the starting point. 

Yep, it sounds a good idea to have programmability via PCIe. To support it, apart from the hardware part, you also need to implement software to control it. You can start a new thread related to this.

qianyich, 2024-04-15:
@zhguanw-amd There is a new problem now.

What I did:
1.  I ran read, write, and send_recv successfully with no issues. No error message from dmesg.
2.  I failed to run the network systolic array example, I got the following.

192.100.52.1:

```
sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./network_systolic_mm -d /dev/reconic-mm -p /sys/bus/pci/devices/0000\:3b\:00.0
/resource2 -r 192.100.52.1 -i 192.100.51.1 -u 22222 -t 11111 --dst_qp 2 -c 2>&1 | tee client_debug.log
Info: Device - /dev/reconic-mm
Info: PCIe resource file: /sys/bus/pci/devices/0000:3b:00.0/resource2
src_ip_str = 192.100.52.1
dst_ip_str = 192.100.51.1
Info: mac_addr_t = 00:0a:35:24:44:8a
Info: src_ip = 192.100.52.1
Info: Found network interface: enp59s0
Info: mac_addr_t = 00:0a:35:71:f4:5c
Info: Creating rn_dev
create_rn_dev - testing2
Info: pre-allocated hugepage buffer vir addr = 0x7fb423a00000, physical addr = 0x2f19a00000
Info: Configuring 8 windows in QDMA AXI bridge BDF, each has 128GB mapping
Info: CREATE RDMA DEVICE
Info: OPEN RDMA DEVICE
Info: RDMA global control status registers are configured.
Info: rdma_dev opened
Info: ALLOCATE PD
Info: OPEN DEVICE FILE
Info: ALLOCATE RDMA QP
Allocating qp->sq
Allocating qp->cq
Allocating qp->rq
Info: queue pair setting is done! Configuring RDMA per-queu CSR registers
Info: allocate_rdma_qp - Successfully allocated a rdma qp
Info: CONFIGURE PSN
[Register] RN_RDMA_QCSR_LSTRQREQi=0x60344, qpid=2, value=0xa000abc
[Register] RN_RDMA_QCSR_SQPSNi=0x60340, qpid=2, value=0xabd
Info: generating matrix data, matrix_size = 256
Info: Software Matrix Data generated
Info: Client is connecting to a remote server
Info: Client is connected to a remote server
Info: client received remote offset of A = 0xa350000000000000
Info: client received remote offset of B = 0xa350000000000400
Info: creating an RDMA read WQE for getting Array A
Successfully sent an RDMA read operation for Array A!
Info: Dump register values for debug purpose
Info: [RN_RDMA_GCSR_ERRBUFWPTR      = 0x6006c] = 0x0
Info: [RN_RDMA_GCSR_IPKTERRQWPTR    = 0x60094] = 0x0
Info: [RN_RDMA_GCSR_INSRRPKTCNT     = 0x60100] = 0x1
Info: [RN_RDMA_GCSR_INAMPKTCNT      = 0x60104] = 0x1
Info: [RN_RDMA_GCSR_OUTIOPKTCNT     = 0x60108] = 0x10004
Info: [RN_RDMA_GCSR_OUTAMPKTCNT     = 0x6010c] = 0x1
Info: [RN_RDMA_GCSR_LSTINPKT        = 0x60110] = 0xabd0210
Info: [RN_RDMA_GCSR_LSTOUTPKT       = 0x60114] = 0x157a204
Info: [RN_RDMA_GCSR_ININVDUPCNT     = 0x60118] = 0x0
Info: [RN_RDMA_GCSR_INNCKPKTSTS     = 0x6011c] = 0x0
Info: [RN_RDMA_GCSR_OUTRNRPKTSTS    = 0x60120] = 0x0
Info: [RN_RDMA_GCSR_WQEPROCSTS      = 0x60124] = 0xc0c2040
Info: [RN_RDMA_GCSR_QPMSTS          = 0x6012c] = 0x20002
Info: [RN_RDMA_GCSR_INALLDRPPKTCNT  = 0x60130] = 0x40000
Info: [RN_RDMA_GCSR_INNAKPKTCNT     = 0x60134] = 0x0
Info: [RN_RDMA_GCSR_OUTNAKPKTCNT    = 0x60138] = 0x0
Info: [RN_RDMA_GCSR_RESPHNDSTS      = 0x6013c] = 0x10f02
Info: [RN_RDMA_GCSR_RETRYCNTSTS     = 0x60140] = 0x0
Info: [RN_RDMA_GCSR_INCNPPKTCNT     = 0x60174] = 0x0
Info: [RN_RDMA_GCSR_OUTCNPPKTCNT    = 0x60178] = 0x0
Info: [RN_RDMA_GCSR_OUTRDRSPPKTCNT  = 0x6017c] = 0x1
Info: [RN_RDMA_GCSR_INTSTS          = 0x60184] = 0x10
Info: [RN_RDMA_GCSR_RQINTSTS1       = 0x60190] = 0x0
[... 292 lines cut ...]
[  700.598945] GLBL_DSC_DAT1_DSC_SPC_C2H                [15, 8]   0x0
[  700.598945] GLBL_DSC_DAT1_DSC_SPC_H2C                [ 7, 0]   0x0
[  700.598949] GLBL_DSC_ERR_LOG2                        0x27c     0x10001    65537
[  700.598950] GLBL_DSC_ERR_LOG2_OLD_PIDX               [31,16]   0x1
[  700.598950] GLBL_DSC_ERR_LOG2_NEW_PIDX               [15, 0]   0x1
[  700.598951] eqdma_hw_error_process detected DMA engine error
```

Then I run the dma test on 192.100.52.1:
```
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
/dev/reconic-mm, W off 0x0, 0x3e80000 failed -1.
write file: Input/output error
qianyich@pc167:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
/dev/reconic-mm, read off 0x0 + 0x3e80000 failed -1.
read file: Input/output error
```
dmesg:
```
[  795.624964] onic:qdma_request_wait_for_cmpl: qdma3b000-MM-67: req 0x000000002c539dbf, W,128,0/128,0x0, done 0, err 0, tm 10000.
[  795.636427] onic:qdma_descq_dump: qdma3b000-MM-67: 0x43/0x43, desc sz 1024/1022, pidx 1, cidx 0
[  795.636443] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
```

qianyich, 2024-04-15:
@zhguanw-amd Another problem is after I ran read, write, and send_recv test successfully, I went ahead to test dma on both servers.

dma failed to work on 192.100.51.1:
```
qianyich@pc166:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200
Write scenario
size=65536000 Average BW = 4.884794 GB/sec, average latency = 13416.328467 us
qianyich@pc166:~/RecoNIC/examples/dma_test$ sudo ./dma_test -d /dev/reconic-mm -s 65536000 -c 200 -r
Read scenario
/dev/reconic-mm, read off 0x0 + 0x3e80000 failed -1.
read file: Input/output error
```
dmesg:
```
[  492.857842] onic:error_intr_handler: Error IRQ fired on Funtion#0: index=7, vector=272
[  492.857849] eqdma_hw_error_process: Global Err Reg(0x248) = 0x4
[  492.857852] addr = 0x00000254 val = 0x00100000
[  492.857858] GLBL_DSC_ERR_STS                         0x254     0x100000   1048576
[  492.857861] GLBL_DSC_ERR_STS_RSVD_1                  [31,26]   0x0
[  492.857862] GLBL_DSC_ERR_STS_PORT_ID                 [   25]   0x0
[  492.857864] GLBL_DSC_ERR_STS_SBE                     [   24]   0x0
[  492.857865] GLBL_DSC_ERR_STS_DBE                     [   23]   0x0
[  492.857867] GLBL_DSC_ERR_STS_RQ_CANCEL               [   22]   0x0
[  492.857868] GLBL_DSC_ERR_STS_DSC                     [   21]   0x0
[  492.857869] GLBL_DSC_ERR_STS_DMA                     [   20]   0x1
[  492.857870] GLBL_DSC_ERR_STS_FLR_CANCEL              [   19]   0x0
[  492.857872] GLBL_DSC_ERR_STS_RSVD_2                  [18,17]   0x0
[  492.857873] GLBL_DSC_ERR_STS_DAT_POISON              [   16]   0x0
[  492.857875] GLBL_DSC_ERR_STS_TIMEOUT                 [    9]   0x0
[  492.857876] GLBL_DSC_ERR_STS_FLR                     [    8]   0x0
[  492.857877] GLBL_DSC_ERR_STS_TAG                     [    6]   0x0
[  492.857879] GLBL_DSC_ERR_STS_ADDR                    [    5]   0x0
[  492.857880] GLBL_DSC_ERR_STS_PARAM                   [    4]   0x0
[  492.857881] GLBL_DSC_ERR_STS_BCNT                    [    3]   0x0
[  492.857882] GLBL_DSC_ERR_STS_UR_CA                   [    2]   0x0
[  492.857884] GLBL_DSC_ERR_STS_POISON                  [    1]   0x0
[  492.857889] GLBL_DSC_ERR_LOG0                        0x25c     0xc0000041 -1073741759
[  492.857890] GLBL_DSC_ERR_LOG0_VALID                  [   31]   0x1
[  492.857892] GLBL_DSC_ERR_LOG0_SEL                    [   30]   0x1
[  492.857893] GLBL_DSC_ERR_LOG0_RSVD_1                 [29,13]   0x0
[  492.857895] GLBL_DSC_ERR_LOG0_QID                    [12, 0]   0x41
[  492.857900] GLBL_DSC_ERR_LOG1                        0x260     0x1014     4116
[  492.857902] GLBL_DSC_ERR_LOG1_RSVD_1                 [31,28]   0x0
[  492.857903] GLBL_DSC_ERR_LOG1_CIDX                   [27,12]   0x1
[  492.857905] GLBL_DSC_ERR_LOG1_RSVD_2                 [11, 9]   0x0
[  492.857906] GLBL_DSC_ERR_LOG1_SUB_TYPE               [ 8, 5]   0x0
[  492.857908] GLBL_DSC_ERR_LOG1_ERR_TYPE               [ 4, 0]   0x14
[  492.857912] GLBL_DSC_DBG_DAT0                        0x270     0x0        0
[  492.857914] GLBL_DSC_DAT0_RSVD_1                     [31,30]   0x0
[  492.857915] GLBL_DSC_DAT0_CTXT_ARB_DIR               [   29]   0x0
[  492.857917] GLBL_DSC_DAT0_CTXT_ARB_QID               [28,17]   0x0
[  492.857918] GLBL_DSC_DAT0_CTXT_ARB_REQ               [16,12]   0x0
[  492.857920] GLBL_DSC_DAT0_IRQ_FIFO_FL                [   11]   0x0
[  492.857921] GLBL_DSC_DAT0_TMSTALL                    [   10]   0x0
[  492.857922] GLBL_DSC_DAT0_RRQ_STALL                  [ 9, 8]   0x0
[  492.857924] GLBL_DSC_DAT0_RCP_FIFO_SPC_STALL         [ 7, 6]   0x0
[  492.857926] GLBL_DSC_DAT0_RRQ_FIFO_SPC_STALL         [ 5, 4]   0x0
[  492.857927] GLBL_DSC_DAT0_FAB_MRKR_RSP_STALL         [ 3, 2]   0x0
[  492.857928] GLBL_DSC_DAT0_DSC_OUT_STALL              [ 1, 0]   0x0
[  492.857933] GLBL_DSC_DBG_DAT1                        0x274     0x0        0
[  492.857934] GLBL_DSC_DAT1_RSVD_1                     [31,28]   0x0
[  492.857936] GLBL_DSC_DAT1_EVT_SPC_C2H                [27,22]   0x0
[  492.857937] GLBL_DSC_DAT1_EVT_SP_H2C                 [21,16]   0x0
[  492.857939] GLBL_DSC_DAT1_DSC_SPC_C2H                [15, 8]   0x0
[  492.857940] GLBL_DSC_DAT1_DSC_SPC_H2C                [ 7, 0]   0x0
[  492.857945] GLBL_DSC_ERR_LOG2                        0x27c     0x3ff03ff  67044351
[  492.857946] GLBL_DSC_ERR_LOG2_OLD_PIDX               [31,16]   0x3ff
[  492.857948] GLBL_DSC_ERR_LOG2_NEW_PIDX               [15, 0]   0x3ff
[  492.857949] eqdma_hw_error_process detected DMA engine error
[  503.005208] onic:qdma_request_wait_for_cmpl: qdma3b000-MM-65: req 0x00000000bf355690, R,4194304,0/65536000,0x0, done 0, err 0, tm 10000.
[  503.017464] onic:qdma_descq_dump: qdma3b000-MM-65: 0x41/0x41, desc sz 1024/0, pidx 0, cidx 1
[  503.018028] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  512.944103] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
```

qianyich, 2024-04-15:
@zhguanw-amd  Also sometimes when I insert the onic.ko module, I have the following errors from dmesg. This happens sometimes and a reboot helps resolve this issue, but not sure why this happens randomly. I just report this, and it could be related to the aforementioned QDMA error.

```
[  905.919063] qdma_is_config_bar: Invalid config bar, err:-4
[  905.924560] qdma_hw_access_init: config bar passed is INVALID, err:-1
[  905.931044] onic 0000:3b:00.0: onic_qdma_setup: qdma_device_open() failed: Error Code: -22
[  905.939304] onic 0000:3b:00.0: onic_pci_probe: onic_qdma_setup() failed with status -22
[  905.947330] onic: probe of 0000:3b:00.0 failed with error -22
```

zhguanw-amd, 2024-04-16:
@qianyich did you apply the new onic.patch to RecoNIC/drivers/onic-driver? It seems you're probably working on an old driver. I can run network_systolic_mm without issue. But there is seg fault issue when freeing some buffers and it has nothing to do with qdma and rdma. I will push a fix for this seg fault.

Can you take a screenshot of Line 439 - 453 in onic_main.c?

qianyich, 2024-04-16:
@zhguanw-amd The new onic.patch has been applied.

[image]

The problem is not directly related to network_systolic_mm. After I ran the read, write, and send_recv tests with no issues, the DMA test failed on 192.100.51.1. I will run those tests again tomorrow.

qianyich, 2024-04-16:
I will reprogram the board and run network_systolic_mm first tomorrow to see if that works, and then run those verb tests, and finally the dma test.

zhguanw-amd, 2024-04-16:
Yes, please reprogram the board and make sure it has the fresh environment.

zhguanw-amd, 2024-04-16:
> @zhguanw-amd The new onic.patch has been applied.
> 
> ![image](https://private-user-images.githubusercontent.com/113211875/322674622-112e0eaa-b44e-4706-805a-7f330d7066b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyNzQ2MDYsIm5iZiI6MTcxMzI3NDMwNiwicGF0aCI6Ii8xMTMyMTE4NzUvMzIyNjc0NjIyLTExMmUwZWFhLWI0NGUtNDcwNi04MDVhLTdmMzMwZDcwNjZiOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxNlQxMzMxNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZGZmMzQxNjk0YzQ2MjE5Yzk0YTIyNTY1MWQwMGVkZmVjMTYxMzJhMTkwZjlmMmU3NWE0NWRkODJiYTVlMmJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2LMlKpvAJ-J7-NS3HTn2fkDNGfmBQNEsyHc3-5G_WX0)
> 
> The problem is not directly related to network_systolic_mm. After I ran the read, write, and send_recv tests with no issues, the DMA test failed on 192.100.51.1 (this is what I had after I rebooted both machines and reinserted the kernel module, I did not run the network systolic array example that time). I will run those tests again tomorrow.

Could you comment out Line 12 and 14 in your screenshot above and only leave Line 13, and have a try?

qianyich, 2024-04-16:
@zhguanw-amd I commented out the lines you mentioned, and I ran network_systolic_mm first, it showed Test Passed. And then I ran read, write, and send_recv tests, and they all worked with no issues.  After these tests, I tried to run network_systolic_mm example again, and it failed.

dmesg on 192.100.52.1, maybe this is the seg fault you mentioned before?:
```
[  952.640278] network_systoli[15528]: segfault at 122000000de ip 00007f864e53d94d sp 00007ffffa01f3a0 error 4 in libc-2.27.so[7f864e4a6000+1e7000]
[ 1054.218446] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1140.048294] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1268.669567] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1306.199807] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1306.199813] network_systoli[15626]: segfault at 122000000de ip 00007fbda6a2b94d sp 00007ffe9986c260 error 4 in libc-2.27.so[7fbda6994000+1e7000]
```

But this somehow has affected QDMA. The DMA tests failed as well. It is an error that is similar to the one I had yesterday.
```
[ 1355.762811] onic:qdma_request_wait_for_cmpl: qdma3b000-MM-67: req 0x00000000a25c3472, W,65536000,0/65536000,0x0, done 0, err 0, tm 10000.
[ 1355.775149] onic:qdma_descq_dump: qdma3b000-MM-67: 0x43/0x43, desc sz 1024/895, pidx 640, cidx 512
[ 1355.775478] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1358.805712] onic:error_intr_handler: Error IRQ fired on Funtion#0: index=7, vector=272
[ 1358.805717] eqdma_hw_error_process: Global Err Reg(0x248) = 0x4
[ 1358.805719] addr = 0x00000254 val = 0x00100000
[ 1358.805723] GLBL_DSC_ERR_STS                         0x254     0x100000   1048576
[ 1358.805724] GLBL_DSC_ERR_STS_RSVD_1                  [31,26]   0x0
[ 1358.805725] GLBL_DSC_ERR_STS_PORT_ID                 [   25]   0x0
[ 1358.805726] GLBL_DSC_ERR_STS_SBE                     [   24]   0x0
[ 1358.805726] GLBL_DSC_ERR_STS_DBE                     [   23]   0x0
[ 1358.805727] GLBL_DSC_ERR_STS_RQ_CANCEL               [   22]   0x0
[ 1358.805727] GLBL_DSC_ERR_STS_DSC                     [   21]   0x0
[ 1358.805727] GLBL_DSC_ERR_STS_DMA                     [   20]   0x1
[ 1358.805728] GLBL_DSC_ERR_STS_FLR_CANCEL              [   19]   0x0
[ 1358.805729] GLBL_DSC_ERR_STS_RSVD_2                  [18,17]   0x0
[ 1358.805729] GLBL_DSC_ERR_STS_DAT_POISON              [   16]   0x0
[ 1358.805729] GLBL_DSC_ERR_STS_TIMEOUT                 [    9]   0x0
[ 1358.805730] GLBL_DSC_ERR_STS_FLR                     [    8]   0x0
[ 1358.805730] GLBL_DSC_ERR_STS_TAG                     [    6]   0x0
[ 1358.805731] GLBL_DSC_ERR_STS_ADDR                    [    5]   0x0
[ 1358.805731] GLBL_DSC_ERR_STS_PARAM                   [    4]   0x0
[ 1358.805732] GLBL_DSC_ERR_STS_BCNT                    [    3]   0x0
[ 1358.805732] GLBL_DSC_ERR_STS_UR_CA                   [    2]   0x0
[ 1358.805732] GLBL_DSC_ERR_STS_POISON                  [    1]   0x0
[ 1358.805736] GLBL_DSC_ERR_LOG0                        0x25c     0xc0000043 -1073741757
[ 1358.805737] GLBL_DSC_ERR_LOG0_VALID                  [   31]   0x1
[ 1358.805737] GLBL_DSC_ERR_LOG0_SEL                    [   30]   0x1
[ 1358.805738] GLBL_DSC_ERR_LOG0_RSVD_1                 [29,13]   0x0
[ 1358.805738] GLBL_DSC_ERR_LOG0_QID                    [12, 0]   0x43
[ 1358.805742] GLBL_DSC_ERR_LOG1                        0x260     0x280014   2621460
[ 1358.805742] GLBL_DSC_ERR_LOG1_RSVD_1                 [31,28]   0x0
[ 1358.805743] GLBL_DSC_ERR_LOG1_CIDX                   [27,12]   0x280
[ 1358.805743] GLBL_DSC_ERR_LOG1_RSVD_2                 [11, 9]   0x0
[ 1358.805744] GLBL_DSC_ERR_LOG1_SUB_TYPE               [ 8, 5]   0x0
[ 1358.805744] GLBL_DSC_ERR_LOG1_ERR_TYPE               [ 4, 0]   0x14
[ 1358.805748] GLBL_DSC_DBG_DAT0                        0x270     0x0        0
[ 1358.805748] GLBL_DSC_DAT0_RSVD_1                     [31,30]   0x0
[ 1358.805749] GLBL_DSC_DAT0_CTXT_ARB_DIR               [   29]   0x0
[ 1358.805749] GLBL_DSC_DAT0_CTXT_ARB_QID               [28,17]   0x0
[ 1358.805749] GLBL_DSC_DAT0_CTXT_ARB_REQ               [16,12]   0x0
[ 1358.805750] GLBL_DSC_DAT0_IRQ_FIFO_FL                [   11]   0x0
[ 1358.805750] GLBL_DSC_DAT0_TMSTALL                    [   10]   0x0
[ 1358.805751] GLBL_DSC_DAT0_RRQ_STALL                  [ 9, 8]   0x0
[ 1358.805752] GLBL_DSC_DAT0_RCP_FIFO_SPC_STALL         [ 7, 6]   0x0
[ 1358.805752] GLBL_DSC_DAT0_RRQ_FIFO_SPC_STALL         [ 5, 4]   0x0
[ 1358.805753] GLBL_DSC_DAT0_FAB_MRKR_RSP_STALL         [ 3, 2]   0x0
[ 1358.805753] GLBL_DSC_DAT0_DSC_OUT_STALL              [ 1, 0]   0x0
[ 1358.805757] GLBL_DSC_DBG_DAT1                        0x274     0x0        0
[ 1358.805757] GLBL_DSC_DAT1_RSVD_1                     [31,28]   0x0
[ 1358.805758] GLBL_DSC_DAT1_EVT_SPC_C2H                [27,22]   0x0
[ 1358.805759] GLBL_DSC_DAT1_EVT_SP_H2C                 [21,16]   0x0
[ 1358.805759] GLBL_DSC_DAT1_DSC_SPC_C2H                [15, 8]   0x0
[ 1358.805760] GLBL_DSC_DAT1_DSC_SPC_H2C                [ 7, 0]   0x0
[ 1358.805763] GLBL_DSC_ERR_LOG2                        0x27c     0x2800280  41943680
[ 1358.805764] GLBL_DSC_ERR_LOG2_OLD_PIDX               [31,16]   0x280
[... 146 lines cut ...]
[ 1358.846878] GLBL_DSC_DAT0_RSVD_1                     [31,30]   0x0
[ 1358.846879] GLBL_DSC_DAT0_CTXT_ARB_DIR               [   29]   0x0
[ 1358.846879] GLBL_DSC_DAT0_CTXT_ARB_QID               [28,17]   0x0
[ 1358.846880] GLBL_DSC_DAT0_CTXT_ARB_REQ               [16,12]   0x0
[... cut ...]
[ 1358.846894] GLBL_DSC_ERR_LOG2_OLD_PIDX               [31,16]   0x280
[ 1358.846894] GLBL_DSC_ERR_LOG2_NEW_PIDX               [15, 0]   0x280
[ 1358.846895] eqdma_hw_error_process detected DMA engine error
[ 1369.075104] onic:qdma_request_wait_for_cmpl: qdma3b000-MM-67: req 0x000000004a0a43c4, R,4190208,0/65536000,0x0, done 0, err 0, tm 10000.
[ 1369.087350] onic:qdma_descq_dump: qdma3b000-MM-67: 0x43/0x43, desc sz 1024/0, pidx 639, cidx 640
[ 1369.087674] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
```

zhguanw-amd, 2024-04-16:
I push a fix for that seg fault. You can get a new version network_systolic_mm

qianyich, 2024-04-16:
@zhguanw-amd so the problem in network_systolic_mm has been resolved. I ran network_systolic_mm and read test back and forth. first network_systolic_mm, then read test, and again network_systolic_mm, and finally read test with no issues.

Right after, I ran a write test with no issue, it passed. And the send_recv test also passed. At this point. **I think they are all good, but I ran the read test again, I had a result mismatch error, and the DMA engine broke again.** **There are no error messages from dmesg until I run the DMA test, it just the last read test failed with result mismatch. So the bug is either in the write test or send_recv test.**

dmesg:
```
[  516.463259] IPv6: ADDRCONF(NETDEV_UP): enp59s0: link is not ready
[  516.463268] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0: link becomes ready
[  644.544157] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  650.451113] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  704.906296] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  740.446916] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  760.696267] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  846.452736] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  894.941990] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[  915.548327] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1034.975745] onic:qdma_request_wait_for_cmpl: qdma3b000-MM-67: req 0x00000000eececbd4, W,65536000,0/65536000,0x0, done 0, err 0, tm 10000.
[ 1034.988096] onic:qdma_descq_dump: qdma3b000-MM-67: 0x43/0x43, desc sz 1024/1022, pidx 641, cidx 640
[ 1034.988429] onic 0000:3b:00.0: reconic-mm: Close onic_cdev.
[ 1037.802323] onic:error_intr_handler: Error IRQ fired on Funtion#0: index=7, vector=272
[ 1037.802330] eqdma_hw_error_process: Global Err Reg(0x248) = 0x4
[ 1037.802332] addr = 0x00000254 val = 0x00100000
[ 1037.802339] GLBL_DSC_ERR_STS                         0x254     0x100000   1048576
[ 1037.802341] GLBL_DSC_ERR_STS_RSVD_1                  [31,26]   0x0
[ 1037.802343] GLBL_DSC_ERR_STS_PORT_ID                 [   25]   0x0
[ 1037.802345] GLBL_DSC_ERR_STS_SBE                     [   24]   0x0
[ 1037.802346] GLBL_DSC_ERR_STS_DBE                     [   23]   0x0
[ 1037.802347] GLBL_DSC_ERR_STS_RQ_CANCEL               [   22]   0x0
[ 1037.802348] GLBL_DSC_ERR_STS_DSC                     [   21]   0x0
[ 1037.802350] GLBL_DSC_ERR_STS_DMA                     [   20]   0x1
[ 1037.802351] GLBL_DSC_ERR_STS_FLR_CANCEL              [   19]   0x0
[ 1037.802353] GLBL_DSC_ERR_STS_RSVD_2                  [18,17]   0x0
[ 1037.802354] GLBL_DSC_ERR_STS_DAT_POISON              [   16]   0x0
[ 1037.802355] GLBL_DSC_ERR_STS_TIMEOUT                 [    9]   0x0
[ 1037.802357] GLBL_DSC_ERR_STS_FLR                     [    8]   0x0
[ 1037.802358] GLBL_DSC_ERR_STS_TAG                     [    6]   0x0
[ 1037.802359] GLBL_DSC_ERR_STS_ADDR                    [    5]   0x0
[ 1037.802361] GLBL_DSC_ERR_STS_PARAM                   [    4]   0x0
[ 1037.802362] GLBL_DSC_ERR_STS_BCNT                    [    3]   0x0
[ 1037.802363] GLBL_DSC_ERR_STS_UR_CA                   [    2]   0x0
[ 1037.802364] GLBL_DSC_ERR_STS_POISON                  [    1]   0x0
[ 1037.802370] GLBL_DSC_ERR_LOG0                        0x25c     0xc0000043 -1073741757
[ 1037.802371] GLBL_DSC_ERR_LOG0_VALID                  [   31]   0x1
[ 1037.802373] GLBL_DSC_ERR_LOG0_SEL                    [   30]   0x1
[ 1037.802374] GLBL_DSC_ERR_LOG0_RSVD_1                 [29,13]   0x0
[ 1037.802376] GLBL_DSC_ERR_LOG0_QID                    [12, 0]   0x43
[ 1037.802381] GLBL_DSC_ERR_LOG1                        0x260     0x280014   2621460
[ 1037.802382] GLBL_DSC_ERR_LOG1_RSVD_1                 [31,28]   0x0
[ 1037.802384] GLBL_DSC_ERR_LOG1_CIDX                   [27,12]   0x280
[ 1037.802385] GLBL_DSC_ERR_LOG1_RSVD_2                 [11, 9]   0x0
[ 1037.802387] GLBL_DSC_ERR_LOG1_SUB_TYPE               [ 8, 5]   0x0
[ 1037.802389] GLBL_DSC_ERR_LOG1_ERR_TYPE               [ 4, 0]   0x14
[ 1037.802393] GLBL_DSC_DBG_DAT0                        0x270     0x0        0
[ 1037.802395] GLBL_DSC_DAT0_RSVD_1                     [31,30]   0x0
[ 1037.802396] GLBL_DSC_DAT0_CTXT_ARB_DIR               [   29]   0x0
[ 1037.802398] GLBL_DSC_DAT0_CTXT_ARB_QID               [28,17]   0x0
[ 1037.802399] GLBL_DSC_DAT0_CTXT_ARB_REQ               [16,12]   0x0
[ 1037.802400] GLBL_DSC_DAT0_IRQ_FIFO_FL                [   11]   0x0
[ 1037.802402] GLBL_DSC_DAT0_TMSTALL                    [   10]   0x0
[ 1037.802403] GLBL_DSC_DAT0_RRQ_STALL                  [ 9, 8]   0x0
[ 1037.802405] GLBL_DSC_DAT0_RCP_FIFO_SPC_STALL         [ 7, 6]   0x0
[ 1037.802406] GLBL_DSC_DAT0_RRQ_FIFO_SPC_STALL         [ 5, 4]   0x0
[ 1037.802408] GLBL_DSC_DAT0_FAB_MRKR_RSP_STALL         [ 3, 2]   0x0
[ 1037.802409] GLBL_DSC_DAT0_DSC_OUT_STALL              [ 1, 0]   0x0
[ 1037.802413] GLBL_DSC_DBG_DAT1                        0x274     0x0        0
[ 1037.802415] GLBL_DSC_DAT1_RSVD_1                     [31,28]   0x0
[ 1037.802416] GLBL_DSC_DAT1_EVT_SPC_C2H                [27,22]   0x0
[ 1037.802418] GLBL_DSC_DAT1_EVT_SP_H2C                 [21,16]   0x0
[ 1037.802420] GLBL_DSC_DAT1_DSC_SPC_C2H                [15, 8]   0x0
[ 1037.802421] GLBL_DSC_DAT1_DSC_SPC_H2C                [ 7, 0]   0x0
[... 385 lines cut ...]
Info: [RN_RDMA_GCSR_CQINTSTS6       = 0x601c4] = 0x0
Info: [RN_RDMA_GCSR_CQINTSTS7       = 0x601c8] = 0x0
Info: [RN_RDMA_GCSR_CQINTSTS8       = 0x601cc] = 0x0
Info: [RN_RDMA_QCSR_CQHEADi         = 0x60330] = 0x1
[... cut ...]
Error: received data mismatched: recv[0]=541065216, sw_golden[0]=0
/users/qianyich/RecoNIC/lib/rdma_api.c:1088:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f93df1fa000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/users/qianyich/RecoNIC/lib/rdma_api.c:1093:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f93df1fa000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/users/qianyich/RecoNIC/lib/rdma_api.c:1101:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f93df1fa000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
```

qianyich, 2024-04-16:
@zhguanw-amd I think, at this point, it is just some bugs in the test programs. RecoNIC is a bit fragile, and any small mistakes in the code could trap the device in an erroneous state, and the pain is that it can never be recovered unless we reprogram the board. BTW, I think the hardware design is ok on U280. We can merge that into the repo if you want.

qianyich, 2024-04-16:
Commenting out the if block in onic_main.c did work. Just let you know if you want to change it in the main branch.

zhguanw-amd, 2024-04-17:
@qianyich QP fatal recovery is a bug in the rdma IP, which is fixed in 4.0 version. I'll push this newer version when I have time.

The above QDMA issue is related to QDMA MM and ST channels mapping. You can revert back to the onic-driver in commit "9e4f0b74bc69744d6d807115e9a23705ba967dbb". But in this commit, there will be around 2-3% ping packet loss due to pid assigned to netdev exceeding 64 occasionally. The latter pushes are used to fix this isse, but seems introducing other qdma problems.

qianyich, 2024-04-17:
@zhguanw-amd Should I just wait for the new release?

I found that QPN, PSN, and rkey are hard coded in the tests. Does RecoNIC provide APIs that generate QPN, rkey, PSN, etc? RNICs from other vendors usually have their own algorithms to generate QPN, etc. Otherwise, it could be insecure, although it is still insecure with some algorithms.

zhguanw-amd, 2024-04-18:
@qianyich For the public release with new RDMA IP, it might be around June or July this year, as we need to upgrade Vivado version to support the new IP. And we also need to change current QDMA due to Vivado upgrade.

Regarding QPN, PSN and rkey, what we provide in the tests is just a showcase to demonstrate how to use RecoNIC and its RDMA via user-space APIs. Designers or users can change it according to their requirements. For example, security concerns you have. I'll leave this to developers if using this libreconic.

More standard usage would be to go through RDMA-core library, which abstracts those variables from users. We have a version at the moment. But for public release, it would take longer time.

It seems most of the issues have been addressed. I'm going to close this thread. Feel free to open another threads if you have more questions.

closed 2024-04-22 by zhguanw-amd