Xilinx/RecoNIC #21: Will Xilinx ERNIC 5.0 be released in the near future?

jiandongg, 2024-04-22:
Hello Xilinx developers! Really appreciate your excellent work. ERNIC 4.0 supports 2048 QP pairs, which is not applicable in our product. Will Xilinx update and release ERNIC 5.0 in the near future?

zhguanw-amd, 2024-04-22:
@jiandongg 
Hi Jiandong,

I have no idea about the roadmap of ERNIC, as I'm not from the product team. We'll include ERNIC 4.0 in RecoNIC this year. May I know the reason why ERNIC4.0 is not applicable in your product? It requires more queues?

jiandongg, 2024-04-22:
> @jiandongg Hi Jiandong,
> 
> I have no idea about the roadmap of ERNIC, as I'm not from the product team. We'll include ERNIC 4.0 in RecoNIC this year. May I know the reason why ERNIC4.0 is not applicable in your product? It requires more queues?

Yes. My project manager/teacher investigated RDMA NICs from mellanox or resnics company. He thought that 2048 QPs of ERNIC 4.0 is a little bit low.
I noticed that Xilinx released ERNIC v3.1 on Vivado 2021.1, and released ERNIC v4.0 on Vivado 2022.2. If Xilinx releases ERNIC 5.0 on Vivado 2024.1 this year(wondering, I don't know the roadmap of ERNIC from public sources either), it would be nice :D

jiandongg, 2024-04-22:
> @jiandongg Hi Jiandong,
> 
> I have no idea about the roadmap of ERNIC, as I'm not from the product team. We'll include ERNIC 4.0 in RecoNIC this year. May I know the reason why ERNIC4.0 is not applicable in your product? It requires more queues?