Xilinx/RecoNIC #22: P4 Build Requirement for RecoNIC?

jyoung3131, 2024-07-11:
We've been trying to compile RecoNIC and were under the impression that it required the ErNIC IP but not the P4 IP. We have obtained an ErNIC license and it is active for our install of Vivado/Vitis.

Is this an accurate assumption that we can compile without P4? Is there another step we might be missing?

To reproduce:
- Running on Ubuntu 22.04, Linux kernel 5.15
- Vitis/Vivado 2021.2
- Python 3.10.2
- We also attempted to compile with `export VitisNetP4_Option_VISIBLE=false' (noting the line to set it to true in the README)
- Head of the RecoNIC repo used, `4e58a87`

Command we ran that triggered the error:
```
#Previous commands in the README were run before this step. csim completes correctly.
RecoNIC/scripts$ make PYTHON_EXE=python3 build_nic
```

Error output: 
```
[Wed Jul 10 20:27:43 2024] reconic_axil_crossbar_synth_1 finished
wait_on_runs: Time (s): cpu = 00:01:44 ; elapsed = 00:01:59 . Memory (MB): peak = 4571.645 ; gain = 0.000 ; free physical = 239141 ; free virtual = 260553
WARNING: [Vivado 12-818] No files matched '<USERDIR>/RecoNIC/base_nics/open-nic-shell/build/au250/vivado_ip/packet_parser/packet_parser.xci'
WARNING: [Vivado 12-818] No files matched '<USERDIR>/RecoNIC/base_nics/open-nic-shell/build/au250/vivado_ip/packet_parser/packet_parser.xci'
p4file = <USERDIR>/RecoNIC/base_nics/open-nic-shell/src/box_250mhz/reconic/packet_classification/packet_parser.p4
CRITICAL WARNING: [IP_Flow 19-3432] UI File /tools/xilinx/2021.2/Vivado/2021.2/data/ip/xilinx/vitis_net_p4_v1_0/xgui/vitis_net_p4_v1_0.tcl does not exist
WARNING: [IP_Flow 19-2991] No XGUI Files found
WARNING: [IP_Flow 19-2162] IP 'packet_parser' is locked:
* IP 'packet_parser' requires one or more mandatory licenses but no valid licenses were found. However license checkpoints may prevent use of this IP in some tool flows.
ERROR: [Common 17-107] Cannot change read-only property 'CONFIG.P4_FILE'.
Resolution: Please refer to Vivado Properties Reference Guide (UG912) for more information on setting properties.

    while executing
"source ${ip_tcl_dir}/${ip}.tcl"
    invoked from within
"if {[file exists "${ip_tcl_dir}/${ip}_${board}.tcl"]} {
            source ${ip_tcl_dir}/${ip}_${board}.tcl
        } elseif {[file exists "${ip_tcl_d..."
    ("foreach" body line 30)
    invoked from within
"foreach ip $ips {
        # Pre-save IP name and its build directory to a global dictionary
        dict append ip_dict $ip ${ip_build_dir}/${ip}

   ..."
    ("dict for" body line 17)
    invoked from within
"dict for {module module_dir} $module_dict {
    set ip_tcl_dir ${module_dir}/vivado_ip

    # Check the existence of "$ip_tcl_dir" and "${ip_tcl_dir}/..."
    (file "build.tcl" line 225)
INFO: [Common 17-206] Exiting Vivado at Wed Jul 10 20:27:45 2024...
make: *** [Makefile:58: build_nic] Error 1
```

zhguanw-amd, 2024-07-12:
@jyoung3131 
Hi Jeffrey,

Regarding the licenses, you don't need VitisNetP4 license, since it's free and integrated into Vivado. However, by ONLY setting "export VitisNetP4_Option_VISIBLE=false" is NOT enough as I mentioned in the README file. You **need to tick** "Vitis Networking P4" when you install Vivado via "Xilinx Unified Installer". Otherwise, you won't see Vitis Networking P4 in your Vivado "IP Catalog".

jyoung3131, 2024-07-17:
Hi Henry,

Thank you very much - do you know if we can use a newer Vivado than 2021.2 or do we need to stick with that version?

zhguanw-amd, 2024-07-18:
@jyoung3131 
Hi Jeff,

To use newer Vivado version (2022.2 and above), we need to upgrade our (1) QDMA IP (from 4.0 to 5.0), (2) ERNIC IP (from 3.1 to 4.0) and (3) adjust system register space. We have (2) and (3) in our internal version, but still don't have bandwidth for (1). Once we have all the three components updated, we will push to this repository.

closed 2024-08-15 by zhguanw-amd