Xilinx/RecoNIC #24: VM Hangs When Bringing Up Interface with RecoNIC Driver

ertza, 2024-07-30:
### Issue Description

I am experiencing an issue where my VM hangs and stops responding when I attempt to bring up the network interface created by the RecoNIC driver. 

### Steps to Reproduce

1. Compile RecoNIC to get the bit file.
2. Program the FPGA (U250).
3. Compile and install the driver on the VM.
4. Observe the new network interface `ens16` appear.
5. Run `ifconfig ens16 up`.
6. The VM hangs and stops responding.

### Expected Behavior

The `ens16` interface should come up without causing the VM to hang.

### Actual Behavior

The VM hangs and stops responding when `ifconfig ens16 up` is executed.

### System Information

- **OS**: Ubuntu 20.04
- **Kernel Version**: 5.4.0-190-generic
- **FPGA**: U250

### Debugging Steps Taken

1. Ran `watch dmesg` to monitor logs, but no messages are shown before the VM hangs.
2. Recompiled both RecoNIC and the driver, but the issue persists.
3. Tried `ip link set ens16 up` in addition to `ifconfig ens16 up`, but the result was the same.

### Additional Information

- Are there any specific logs or outputs that might help in debugging?
- Has anyone encountered a similar issue?
- Any suggestions for further steps to debug or resolve this problem?

zhguanw-amd, 2024-07-31:
@ertza 
Hi ertza,

We tried RecoNIC with docker setup before and it worked fine. But we haven't tried VM setup with RecoNIC in our testing.

Are you able to dump dmesg logs when you load onic.ko? Without the log file, it's hard to understand which part has issues. For VM configuration, have you done any setup to expose things (like PCIe devices, /dev/ and /sys/kernel/mm/hugepages) from host OS to your VM?

ertza, 2024-07-31:
Hi, I am using Proxmox VE, which is a hypervisor, so the VMs should essentially behave similar to a bare-metal machine. The FPGA is connected as a raw PCIe device to the VM with PCIe passthrough, the hugepages are configured as follows inside the VM:
```
HugePages_Total:     256
HugePages_Free:      256
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:          524288 kB
```

Also attaching the dmesg log:
```
Jul 31 17:19:07 celeris1 kernel: [  411.054442] onic: loading out-of-tree module taints kernel.
Jul 31 17:19:07 celeris1 kernel: [  411.054634] onic: module verification failed: signature and/or required key missing - tainting kernel
Jul 31 17:19:07 celeris1 kernel: [  411.056128] onic 0000:00:10.0 onic0s16f0 (uninitialized): Set MAC address to 0:a:35:64:7:1b
Jul 31 17:19:07 celeris1 kernel: [  411.056130] onic 0000:00:10.0: device is a master PF
Jul 31 17:19:07 celeris1 kernel: [  411.056158] onic_set_num_queue: num_msix 8, nb_queues 7, pci_msix_user_cnt 1
Jul 31 17:19:07 celeris1 kernel: [  411.056203] real_num_tx_queues: 64 real_num_rx_queues 64
Jul 31 17:19:07 celeris1 kernel: [  411.056204] onic:qdma_device_open: onic, 00:10.00, pdev 0x00000000f0e6f7ad, 0x10ee:0x903f.
Jul 31 17:19:07 celeris1 kernel: [  411.077151] Device Type: Soft IP
Jul 31 17:19:07 celeris1 kernel: [  411.077152] IP Type: EQDMA Soft IP
Jul 31 17:19:07 celeris1 kernel: [  411.077152] Vivado Release: vivado 2020.2
Jul 31 17:19:07 celeris1 kernel: [  411.077158] onic:qdma_device_attributes_get: qdma00100-p0000:00:10.0: num_pfs:1, num_qs:512, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
Jul 31 17:19:07 celeris1 kernel: [  411.077160] onic:qdma_device_open: Vivado version = vivado 2020.2
Jul 31 17:19:07 celeris1 kernel: [  411.077161] qdma_dev_entry_create: Created the dev entry successfully
Jul 31 17:19:07 celeris1 kernel: [  411.103507] onic:xdev_identify_bars: AXI Master Lite BAR 2.
Jul 31 17:19:07 celeris1 kernel: [  411.103509] onic:qdma_device_open: 0000:00:10.0, 00100, pdev 0x00000000f0e6f7ad, xdev 0x00000000a89947b9, ch 1, q 68, vf 0.
Jul 31 17:19:07 celeris1 kernel: [  411.103619] onic 0000:00:10.0: successfully cdev_add a character device, reconic-mm, to the system
Jul 31 17:19:07 celeris1 kernel: [  411.103686] onic 0000:00:10.0: successffully device_create a character device, reconic-mm, and register it
Jul 31 17:19:09 celeris1 kernel: [  412.346751] onic 0000:00:10.0 ens16: renamed from onic0s16f0
```
There is no message after loading the driver, i.e. no message appears when I do `ifconfing ens16 up` and the VM just hangs.

zhguanw-amd, 2024-08-01:
@ertza 

Hi,

I haven't used Proxmox VE. From the log you showed, it seems to be okay. From the behavior you described, the issue might be related to the VM configuration

Instead of using "ifconfig ens16 up", can you also try with netplan detailed in the [README](https://github.com/Xilinx/RecoNIC/blob/main/README.md) file?

ertza, 2024-08-01:
I tried the netplan approach and the result was same, as soon as I ran `sudo netplan apply` the VM hangs. 

Any other ideas to debug this except dmesg? In parallel I will be looking into if I can get my hands on a bare-metal machine for running this.

zhguanw-amd, 2024-08-01:
@ertza probably you can add some debug printout in drivers/onic-driver/onic_main.c to understand which line of code causes this hanging issue. You can check two functions: onic_pci_probe() and onic_open().

ertza, 2024-08-01:
Perfect, that's exactly the kind of starting point I was hoping for. Will try to debug and report back what I find.

ertza, 2024-08-02:
So I was able to get a bare-metal machine and was successful in setting up the RecoNIC driver. Thank you for the support. However I do have a small question, I saw https://github.com/Xilinx/RecoNIC/issues/14 and wanted to confirm if this means that RecoNIC cannot communicate with other nodes having standard RDMA systems e.g. ConnectX-6 RDMA or SoftRoCE etc.? I only have one FPGA at the moment, but to use RecoNIC's RDMA implementation I need RecoNIC (and hence another FPGA) on both sides? I was able to ping other non-RecoNIC systems, so I am assuming only the RDMA implementation is not the standard one?

zhguanw-amd, 2024-08-04:
@ertza 

Hi ertza, ERNIC supports standard RoCEv2 protocol and it can work with ConnectX-6's RDMA. RecoNIC is able to communicate with other nodes having standard RDMA systems.

ertza, 2024-08-05:
Perfect, thank you for confirming!

closed 2024-08-05 by ertza

qianyich, 2024-09-20:
> @ertza
> 
> Hi ertza, ERNIC supports standard RoCEv2 protocol and it can work with ConnectX-6's RDMA. RecoNIC is able to communicate with other nodes having standard RDMA systems.

Hi Guanwen,

Just returned to school recently. May I know if this is true with the current version of RecoNIC or do we need to wait until the use of ERNIC 4.0?