Xilinx/RecoNIC #25: Missing: 'sdnet_p4@2021.06'

aexolate, 2024-08-26:
Hello, I encountered the P4 IP error ( similar to Issue #22 ) when running `vivado -mode batch -source gen_vivado_ip.tcl -tclargs -board_repo $BOARD_REPO`. I used `export VitisNetP4_Option_VISIBLE=true` and ticked it during installation, and I can see it in my IP Catalog as well, but the error message suggests the sdnet_p4@2021.06 license is still required. Is this a licensing issue?

[image]

```
ERROR: [xilinx.com:ip:vitis_net_p4:1.0-0] packet_parser: Attempting to use IP without required license. Missing: 'sdnet_p4@2021.06'.

    while executing
"package require vitisnetp4"
    (file "/tools/Xilinx/Vivado/2021.2/data/ip/xilinx/vitis_net_p4_v1_0/xgui/vitis_net_p4_v1_0.tcl" line 1)
WARNING: [IP_Flow 19-3303] Errors found in sourcing XGUI file '/tools/Xilinx/Vivado/2021.2/data/ip/xilinx/vitis_net_p4_v1_0/xgui/vitis_net_p4_v1_0.tcl'. 
ERROR: [IP_Flow 19-3428] Failed to create Customization object packet_parser
CRITICAL WARNING: [IP_Flow 19-5622] Failed to create IP instance 'packet_parser'. Failed to customize IP instance 'packet_parser'. Failed to load customization data
ERROR: [Common 17-69] Command failed: Create IP failed with errors
```

zhguanw-amd, 2024-09-26:
@aexolate 
Hi,

It seems your installation has some issues. **Vitis_Networking_P4 doesn't need license** in Vivado 2021.2 and above. Could you try Vitis Networking P4 examples and see whether you can use it?

Thanks,
Henry

zhguanw-amd, 2024-10-27:
Close this thread due to inactivity. Feel free to reopen it if you have further questions.

closed 2024-10-27 by zhguanw-amd