Xilinx/RecoNIC #30: make issues when building onic driver

darrylkid, 2024-10-26:
Following the onic driver installation steps [here](https://github.com/Xilinx/RecoNIC?tab=readme-ov-file#:~:text=for%20the%20instructions.-,Driver%20Installation,-Install%20the%20modified), I run into the following issue when trying to `make`:

```bash
  1 make
  2 make -C /lib/modules/4.18.0-553.22.1.el8_10.x86_64/build M=<root path>/RecoNIC/drivers/onic-driver SUBDIRS=<root path>/RecoNIC/drivers/onic-driver modules
  3 make[1]: Entering directory '/usr/src/kernels/4.18.0-553.22.1.el8_10.x86_64'
  4   CC [M]  <root path>/RecoNIC/drivers/onic-driver/./onic_main.o
  5 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c: In function ‘onic_rx_poll’:
  6 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c:254:5: error: implicit declaration of function ‘napi_reschedule’; did you mean ‘napi_schedule’? [-Werror=implicit-function-declarati    on]
  7      napi_reschedule(napi);
  8      ^~~~~~~~~~~~~~~
  9      napi_schedule
 10 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c: In function ‘onic_qdma_rx_queue_setup’:
 11 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c:403:30: error: macro "netif_napi_add" passed 4 arguments, but takes just 3
 12               ONIC_NAPI_WEIGHT);
 13                               ^
 14 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c:402:5: error: ‘netif_napi_add’ undeclared (first use in this function); did you mean ‘netif_napi_del’?
 15      netif_napi_add(xpriv->netdev, &xpriv->napi[q_no], onic_rx_poll,
 16      ^~~~~~~~~~~~~~
 17      netif_napi_del
 18 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c:402:5: note: each undeclared identifier is reported only once for each function it appears in
 19 At top level:
 20 <root path>/RecoNIC/drivers/onic-driver/./onic_main.c:206:12: warning: ‘onic_rx_poll’ defined but not used [-Wunused-function]
 21  static int onic_rx_poll(struct napi_struct *napi, int quota)
 22             ^~~~~~~~~~~~
 23 cc1: some warnings being treated as errors
 24 make[2]: *** [scripts/Makefile.build:318: <root path>/RecoNIC/drivers/onic-driver/./onic_main.o] Error 1
 25 make[1]: *** [Makefile:1619: _module_<root path>/RecoNIC/drivers/onic-driver] Error 2
 26 make[1]: Leaving directory '/usr/src/kernels/4.18.0-553.22.1.el8_10.x86_64'
 27 make: *** [Makefile:24: all] Error 2
```

zhguanw-amd, 2024-10-27:
Hi @darrylkid ,

Your linux kerner version is too old. The error message shows that netif_napi_add has different arguments.

zhguanw-amd, 2024-11-01:
Close this thread due to inactivity. Feel free to reopen it if you have further questions.

closed 2024-11-01 by zhguanw-amd