Xilinx/RecoNIC #32: When I run the simulation command, the following error occurs

qgzln, 2024-11-05:
Vivado Simulator v2021.2
Copyright 1986-1999, 2001-2021 Xilinx, Inc. All Rights Reserved.
Running: /home/soc_qg/software/vitis2021.2/Vivado/2021.2/bin/unwrapped/lnx64.o/xelab --incr --relax --debug typical --mt auto -L reco -L ernic_v3_1_1 -L xilinx_vip -L xpm -L cam_v2_2_2 -L -L axi_protocol_checker_v2_0_8 -L unisims_ver -L unimacro_ver -L secureip --snapshot rn_tb_2rdma_top_opt reco.rn_tb_2rdma_top reco.glbl -log xsim_elaborate.log
Multi-threading is on. Using 94 slave threads.
Starting static elaboration
Pass Through NonSizing Optimizer
ERROR: [VRFC 10-2063] Module <sync_fifo_fg> not found while processing module instance <WR_DATA_BUF.wr_retry_data_buf_fifo_2> [/wrk/ci/prod/2021.2/sw/continuous/8687/packages/customer/vivado/data/ip/xilinx/ernic_v3_1/hdl/ernic_v3_1_rfs.sv:43756]
ERROR: [VRFC 10-2865] module 'ernic_v3_1_1_wqe_proc_hdr_gen(C_MAX_QP=32,C_MAX_QID_WIDTH=5,C_MAX_WRDATA_BUF_NUM=2048,C_OS_Q_INDX_WIDTH=4,C_M_AXI_ADDR_WIDTH=64,C_EN_DEBUG=1,C_EN_RECV_RD_WR=1,NVMOF_SYSTEM_ONLY_FEATURE_SEND_INVALIDATE_EN=1)' ignored due to previous errors [/wrk/ci/prod/2021.2/sw/continuous/8687/packages/customer/vivado/data/ip/xilinx/ernic_v3_1/hdl/ernic_v3_1_rfs.sv:1]
ERROR: [VRFC 10-2865] module 'xpm_fifo_async(WR_DATA_COUNT_WIDTH=12,READ_MODE="fwft",FIFO_READ_LATENCY=0,READ_DATA_WIDTH=32)' ignored due to previous errors [/wrk/ci/prod/2021.2/sw/continuous/8687/packages/customer/vivado/data/ip/xpm/xpm_fifo/hdl/xpm_fifo.sv:2145]
ERROR: [VRFC 10-2865] module 'mmult_systolic_m_axi(NUM_READ_OUTSTANDING=16,NUM_WRITE_OUTSTANDING=16,C_M_AXI_ADDR_WIDTH=64,C_M_AXI_DATA_WIDTH=512,C_USER_VALUE=0,C_PROT_VALUE=0,C_CACHE_VALUE=3,USER_DW=512,USER_AW=64,USER_MAXREQS=5,USER_RFIFONUM_WIDTH=9)' ignored due to previous errors [/home/soc_qg/prj/rdma/RecoNIC/shell/compute/lookside/kernel/mmult_systolic_m_axi.v:8]

I removed P4, instead of  my verilog code , xsim.ini did not change any more when i ran the script

zhguanw-amd, 2024-11-05:
@qgzln The error is caused by "sync_fifo_fg" module not found.

We have tested simulation with xsim in this public repository again by cloning a fresh copy this afternoon. We didn't encounter any issues. The issue should be related to your Vivado environment or installation.

qgzln, 2024-11-05:
@zhguanw-amd The vivado version is 2021.2. "sync_fifo_fg" This IP seems to be part of ernic_v3_1. I removed the P4 part and modified the compile.do file to eliminate the P4 compilation part. I generate all the IP using  gen_vivado_ip.tcl.it did not make any errors and prints All IPs required for simulation are generated

qgzln, 2024-11-05:
> @qgzln The error is caused by "sync_fifo_fg" module not found.
> 
> We have tested simulation with xsim in this public repository again by cloning a fresh copy this afternoon. We didn't encounter any issues. The issue should be related to your Vivado environment or installation.

@zhguanw-amd The vivado version is 2021.2. "sync_fifo_fg" This IP seems to be part of ernic_v3_1. I removed the P4 part and modified the compile.do file to eliminate the P4 compilation part. I generate all the IP using  gen_vivado_ip.tcl.it did not make any errors and prints All IPs required for simulation are generated
[image]

zhguanw-amd, 2024-11-05:
@qgzln We don't have any issues with the xsim/questasim and I'm not able to reproduce your error.

ERNIC uses sync_fifo_fg module, which is a built-in FIFO IP. If your installation is okay, you should be able to find it under "Vivado/2021.2/data/ip/xilinx/lib_fifo_v1_0/hdl/lib_fifo_v1_0_rfs.vhd". In your specific case, you can also try to link this library in simulate.sh in sim/scripts/simulate.sh.

zhguanw-amd, 2024-12-20:
Close this thread due to inactivity. Feel free to reopen it if you have further questions.

closed 2024-12-20 by zhguanw-amd