Xilinx/RecoNIC #34: `dmesg | grep onic` error: No MSIC cpability -22

darrylkid, 2024-12-13:
### Error

```bash
<path>/RecoNIC/drivers/onic-driver$ sudo dmesg | grep onic
[735563.768106] onic 0000:42:00.0 onic66s0f0 (uninitialized): Set MAC address to 0:a:35:8a:6:f4
[735563.768111] onic 0000:42:00.0: device is a master PF
[735563.768112] onic_set_num_queue:No MSIX cpability -22
[735563.768120] onic: probe of 0000:42:00.0 failed with error -22
[735573.388267] onic 0000:42:00.0 onic66s0f0 (uninitialized): Set MAC address to 0:a:35:88:56:5c
[735573.388273] onic 0000:42:00.0: device is a master PF
[735573.388276] onic_set_num_queue:No MSIX cpability -22
[735573.388288] onic: probe of 0000:42:00.0 failed with error -22
```

### System Info

**OS:** 
```bash
<path>/RecoNIC/drivers/onic-driver$ lsb_release -a
LSB Version:    core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
```

**Driver:**
```bash
<path>/RecoNIC/drivers/onic-driver$ lsmod | grep onic
onic                  585728  0
```

**U250:**
```bash
<path>/RecoNIC/drivers/onic-driver$ lspci | grep Xilinx
42:00.0 Processing accelerators: Xilinx Corporation Device d004
```

zhguanw-amd, 2024-12-14:
Hi @darrylkid ,

Please make sure you have programmed the FPGA board with RecoNIC bitstream properly. From the output of "lspci | grep Xilinx", I don't think you program correctly.

zhguanw-amd, 2024-12-22:
Close this thread due to inactivity. Feel free to reopen it if you have further questions.

closed 2024-12-22 by zhguanw-amd