Xilinx/RecoNIC #37: Question on ernic_4 branch

wcferry, 2025-03-04:
Hi Henry,

Great job on this project!

I am wondering - is the ernic_4 branch feature complete, passing verification / regression test?  Or is it still work in progress?
Could you illuminate the major features added in going to ERNIC v4.0?

Thanks kindly
Will Ferry
drut.io

zhguanw-amd, 2025-03-19:
@wcferry 
Hi Ferry,

Sorry for the delay. We are working with Yicheng from Northeastern University to bring up the ernic4.0 in the system. Once it's done, I'll push to this repository. At the moment, you can check out the repository from his group, which is a forked version of RecoNIC: https://github.com/OCT-FPGA/oceania/tree/upgrade/ernic_4.

To port ERNIC4.0, we need to (1) update ernic's new register space; (2) update p4 IP; (3) have to upgrade QDMA from 4.0 to 5.0 due to Vivado version (and upgrade reconic_mm character driver as well). The current simulation with ernic 4.0 is done and we are working on on-board testing. There are still some driver issues on the QDMA side.

Thanks,
Henry

zhguanw-amd, 2025-03-27:
@wcferry 
Hi Ferry,

We'll upgrade RecoNIC to ernic 4.0 and newer version Vivado soon. Please stay tuned for further updates.

Thanks,
Henry