Xilinx/RecoNIC #39: how to resove MIG error in implement phase

jethtxiao, 2025-04-19:
hase 1 Generate And Synthesize MIG Cores
ERROR: [Mig 66-99] Memory Core Error - [ddr4_inst] Either port(s) c0_sys_clk_p, c0_sys_clk_n is/are not placed or un-supported clocking structure/circuit for memory ip instance. Please refer to clocking section of PG150 for supported clocking structures. 
ERROR: [Mig 66-99] Memory Core Error - [ddr4_inst] Port(s) c0_ddr4_ck_c[0],c0_ddr4_ck_t[0],c0_ddr4_adr[0],c0_ddr4_adr[1],c0_ddr4_adr[2],c0_ddr4_adr[3],c0_ddr4_adr[4],c0_ddr4_adr[5],c0_ddr4_adr[6],c0_ddr4_adr[7],c0_ddr4_adr[8],c0_ddr4_adr[9],c0_ddr4_adr[10],c0_ddr4_adr[11],c0_ddr4_adr[12],c0_ddr4_adr[13],c0_ddr4_adr[14],c0_ddr4_adr[15],c0_ddr4_adr[16],c0_ddr4_ba[0],c0_ddr4_ba[1],c0_ddr4_bg[0],c0_sys_clk_n,c0_sys_clk_p,c0_ddr4_bg[1],c0_ddr4_cs_n[0],c0_ddr4_cke[0],c0_ddr4_odt[0],c0_ddr4_act_n,c0_ddr4_reset_n,c0_ddr4_parity,c0_ddr4_dqs_c[0],c0_ddr4_dqs_t[0],c0_ddr4_dqs_c[1],c0_ddr4_dqs_t[1],c0_d

zhguanw-amd, 2025-04-20:
@jethtxiao 

Could you share more information such as vivado version, FPGA board used and log file? It seems you're using either wrong device or vivado version.

jethtxiao, 2025-04-21:
Thank you, this issue has been resolved. I had to manually add the DDR4 pin constraints to the pins.xdc file. Is there a better solution for this? I'm not sure whether the scripts related to the AU250 board were executed correctly.