Xilinx/RecoNIC #44: RDMA tests error for RecoNIC setup with two U55C's

KamilHanna, 2025-08-16:
We are testing RecoNIC setup with two U55C's boards , with an HBM instead of a ddr4. The DMA tests work perfectly fine whereas rdma tests are malfunctioning, here's an output example of an rdma read from both client and server :

Output from Peer 1 (Client side) :
kamilgiorgio@kamilgiorgio-Standard-PC-Q35-ICH9-2009:~/Documents/RecoNIC/examples/rdma_test$ sudo env LD_LIBRARY_PATH=$LD_LIBRARY_PATH ./read -r 192.100.51.1 -i 192.100.52.1 -p /sys/bus/pci/devices/0000:07:00.0/resource2 -z 128 -l host_mem -d /dev/reconic-mm -c -u 22222 -t 11111 --dst_qp 2  -g 2>&1 | tee client_debug.log
src_ip_str = 192.100.51.1
dst_ip_str = 192.100.52.1
Info: mac_addr_t = 00:0a:35:ca:26:34
Info: PCIe resource file: /sys/bus/pci/devices/0000:07:00.0/resource2
Info: QP allocated at: host_mem
Info: Device - /dev/reconic-mm
Info: src_ip = 192.100.51.1
Info: Found network interface: enp7s0
Info: mac_addr_t = 00:0a:35:92:12:ab
Info: Creating rn_dev
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:321:create_rn_dev(): Info: scr(=4)) file open successfully
create_rn_dev - testing2
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4e000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:163:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x4e000000
Info: pre-allocated hugepage buffer vir addr = 0x7f88d7c00000, physical addr = 0x4e000000
Info: Configuring 8 windows in QDMA AXI bridge BDF, each has 128GB mapping
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16420, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16424, bdf_addr_high=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16430, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16440, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16444, bdf_addr_high=0x20
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16450, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16460, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16464, bdf_addr_high=0x40
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16470, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16480, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16484, bdf_addr_high=0x60
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16490, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x164a0, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x164a4, bdf_addr_high=0x80
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x164b0, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x164c0, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x164c4, bdf_addr_high=0xa0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x164d0, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x164e0, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x164e4, bdf_addr_high=0xc0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x164f0, bdf_win_config=0xc2000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:206:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_LSB=0x16500, bdf_addr_low=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:207:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_ADDR_TRANSLATE_ADDR_MSB=0x16504, bdf_addr_high=0xe0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:208:config_rn_dev_axib_bdf(): [BDF] AXIB_BDF_MAP_CONTROL_ADDR=0x16510, bdf_win_config=0xc2000000
Info: CREATE RDMA DEVICE
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4e000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:163:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x4e000000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:240:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7f88d7c00000, physical addr = 4e000000, rn_dev->buffer_offset = 0x200000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:241:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4e200
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:163:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x4e200000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:240:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7f88d7e00000, physical addr = 4e200000, rn_dev->buffer_offset = 0x1200000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:241:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4f200
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:163:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x4f200000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:240:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7f88d8e00000, physical addr = 4f200000, rn_dev->buffer_offset = 0x1202000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:241:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4f202
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:163:get_buffer_paddr(): Info: get_buffer_paddr - Physical address of buffer: 0x4f202000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:240:allocate_rdma_buffer(): Info: allocated host buffer vir addr = 0x7f88d8e02000, physical addr = 4f202000, rn_dev->buffer_offset = 0x1212000
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:241:allocate_rdma_buffer(): Info: allocate_rdma_buffer - successfully allocated rdma host buffer
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:154:get_buffer_paddr(): Info: get_buffer_paddr - Page frame: 0x4f212
/home/kamilgiorgio/Documents/RecoNIC/lib/reconic.c:159:get_buffer_paddr(): Info: get_buffer_paddr - distance from page boundary: 0x0
[... cut ...]
/home/kamilgiorgio/Documents/RecoNIC/lib/rdma_api.c:1087:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f24cfeae000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/rdma_api.c:1092:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f24cfeae000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
/home/kamilgiorgio/Documents/RecoNIC/lib/rdma_api.c:1100:destroy_rdma_qp(): [DEBUG] Destroying dev: 0x7f24cfeae000, RN_RDMA_QCSR_CQHEADi=0x60330, qpid=2, value=0x0
kamilgiorgio@kamilgiorgio-Standard-PC-Q35-ICH9-2009:~/Documents/RecoNIC/examples/rdma_test$

KamilHanna, 2025-09-14:
Not really.. moved on from that into a new "solution" > Hello brother,I saw your log file and encontered the same problem.I tested RecoNIC setup with two U50 boards, with HBM.I don't know if you have solved the problem.

yao111-hao, 2025-09-17:
I used Wireshark for packet analysis and found that during RDMA read/write operations, the opcode sent by the client side via the QSFP28 interface is only 'write'. So, I suspect that the host is not able to properly control the RDMA engine.